neskweek/LightSaberOS #1: Single Ignition/Retraction Issue

772pilot, 2016-03-17:
Lightsaber only powers on / powers off once. After that, the saber becomes unresponsive - buttons do nothing, no sound is played, and LEDs remain off. 

Config menu works as expected from boot. Can be entered/exited multiple times without issue, but cannot be accessed when the saber becomes unresponsive. 

Problem persists on two different Nano clone boards ([these](http://www.amazon.com/OSOYOO-ATMEGA328P-Module-Micro-controller-Arduino/dp/B00UACD13Q/ref=sr_1_1?s=electronics&ie=UTF8&qid=1458249970&sr=1-1&keywords=arduino+nano)). Issue was not encountered the first few times code was compiled/uploaded to Nano. Tried resetting the board with blank sketch and reuploading LSOS - no success.

772pilot, 2016-03-17:
Updated with more info.

neskweek, 2016-03-18:
Problem source found and corrected in v1.0.1 :

Nano clones seems to have a problem with the corde of the first retractation effect : I replace the counter type to uint8_t to int8_t and it now works OK.

Still makes me think those clones have flaws :P

closed 2016-03-18 by neskweek

772pilot, 2016-03-18:
Works like a beauty now! Thank you! Who would've thought that uint8_t would've been so pesky.