neskweek/LightSaberOS #2: Not able to build in Eclipse

DJWing79, 2016-03-17:
So I took a look at the project and there is a red X nest to subdir.mk

when I click on Problems, I get a error I can highlight

@echo 'Building file: $<'

https://www.dropbox.com/s/lxbd7mcqcb7wxvw/Eclipse%20Error.png?dl=0

neskweek, 2016-03-18:
Hey DJwing79
Thanks for having posted you problem here.

I know you already have posted some description on the forum, but even there I didn't catch the whole thing correctly.
Could your repost here all the error message you did receive, please ?
Could you tell me what board you're using (Trinket or Uno or else)?

The error you posted right above is only a consequence of another problem.

DJWing79, 2016-03-18:
I'm using 3.0.0 Toddler and trying to upload to an arduino uno at this time to make sure it at least works for me 

Ok, new error using Eclipse

17:03:34 **** Incremental Build of configuration Release for project LSOS ****
"C:\\Users\\Damien\\Desktop\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"C:\Users\Damien\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.5\cores\arduino" -I"C:\Users\Damien\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.5\variants\standard" -I"C:\Users\Damien\Arduino\libraries\OneButton" -I"C:\Users\Damien\Arduino\libraries\DFPlayer" -I"C:\Users\Damien\Arduino\libraries\I2Cdev" -I"C:\Users\Damien\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.5\libraries\Wire" -I"C:\Users\Damien\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.5\libraries\Wire\utility" -I"C:\Users\Damien\Arduino\libraries\MPU6050" -I"C:\Users\Damien\Arduino\libraries\LinkedList" -I"C:\Users\Damien\Arduino\libraries\EEPROMEx" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
process_begin: CreateProcess(NULL, sh.exe -c "/bin/avr-g++ -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Users\Damien\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.5\cores\arduino -IC:\Users\Damien\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.5\variants\standard -IC:\Users\Damien\Arduino\libraries\OneButton -IC:\Users\Damien\Arduino\libraries\DFPlayer -IC:\Users\Damien\Arduino\libraries\I2Cdev -IC:\Users\Damien\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.5\libraries\Wire -IC:\Users\Damien\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.5\libraries\Wire\utility -IC:\Users\Damien\Arduino\libraries\MPU6050 -IC:\Users\Damien\Arduino\libraries\LinkedList -IC:\Users\Damien\Arduino\libraries\EEPROMEx -MMD -MP -MF.ino.cpp.d -MT.ino.cpp.o -D__IN_ECLIPSE__=1 -x c++ ../.ino.cpp -o .ino.cpp.o -Wall", ...) failed.
make (e=2): The system cannot find the file specified.
subdir.mk:33: recipe for target '.ino.cpp.o' failed

make: *** [.ino.cpp.o] Error 2

neskweek, 2016-03-18:
Are you using eclipse comming from this archive : 
for win64 : [win64.2016-03-11_03-29-38.tar.gz](http://www.baeyens.it/eclipse/download/product/win64.2016-03-11_03-29-38.tar.gz)
for win32 : [win64.2016-03-11_03-29-38.tar.gz] (http://www.baeyens.it/eclipse/download/product/win32.2016-03-11_03-29-38.tar.gz)

DJWing79, 2016-03-19:
win64-.2016-03-11-03-29.tar.gz

DJWing79, 2016-03-22:
Ok, using your file and following your instructions again, I got it to compile and upload. It works...first DFPlayer fried though, but second one worked fine (was powering the first on 7.4v)

neskweek, 2016-03-22:
Glad to hear you made it work
R.I.P. litlle DFPlayer :(

closed 2016-03-22 by neskweek