neskweek/LightSaberOS #3: Power Alert and Accent

DJWing79, 2016-03-23:
I can't figure out the correct way to enable the accent led. I uncomment the #define ACCENT_LED line and hook the led line to A2 and I still get nothing.

Also, how do I disable the low power warning? I want to use this with a 3.7V battery setup.

neskweek, 2016-03-23:
Hello

Since you're using an RGB LED you should put your accent led on a free PWM pin.
And comment in Con,fig.h
`#define SOFT_ACCENT`

Anyway it should still work on A2 with HARD_ACCENT.     
Are you sure you didn't fryed your led ?      
Some LEDs doesn't accept more than 2,5 V

Concerning low power setup you have to change in LightSaber.ino : 
```C++
/*
 * BATTERY SAFETY MONITOR
 * Modify to match your baterries specs
 * rule of thumbs for serial wired batteries:
 * LOW_BATTERY = 3200 * Nb_of_battery_cells
 * CRITICAL_BATTERY = 3050 * Nb_of_battery_cells
 *************************************/
#define LOW_BATTERY 		3200
#define CRITICAL_BATTERY 	3050
/************************************/

```
But they already are set for a single 3.7V battery. If you don't have any sound when plugging your battery, then the problem doesn't come form the code.

DJWing79, 2016-03-24:
Ok, I got the accent to work, I had to // on the SOFT_ACCENT, when I did HARD_ACCENT it gave me an error. But it flashes really fast, is there a way to slow the pulsing down when it is not in running mode?

neskweek, 2016-03-24:
In LightSaber.ino : 

```C++
#if defined HARD_ACCENT
		if (millis() - lastAccent <= 400) {
			analogWrite(ACCENT_LED, millis() - lastAccent);
		} else if (millis() - lastAccent > 400
				and millis() - lastAccent <= 800) {
			analogWrite(ACCENT_LED, 800 - (millis() - lastAccent));
		} else {
			lastAccent = millis();
		}
#endif
```
Changing 400 and 800 by augmenting them will slow down the blink.

DJWing79, 2016-03-24:
Ok, I moved the following pins and it will turn on but then won't turn off

MAIN_BUTTON 8
LOCKUP_BUTTON 4
DFPLAYER_RX A2
DFPLAYER_TX A3
ACCENT_LED 6
LED_RED 5
LED_GREEN 10
LED_BLUE 9
ATTACHINTERRUPT 1 (moving it to pin 3)

Also, whenever it boots it plays a part of the hum sound, is that by design? I was shown how to do a boot sound in my other sketches, just wondering if this is intentional.

DJWing79, 2016-03-25:
Here's the Config.h

/*
 * Config.h
 *
 * Created on: 6 mars 2016
 * author: 		Sebastien CAPOU (neskweek@gmail.com)
 * Source : 	https://github.com/neskweek/LightSaberOS
 */

#ifndef CONFIG_H_
#define CONFIG_H_

/*!!!!!IMPORTANT IMPORTANT IMPORTANT IMPORTANT IMPORTANT!!!
 *
 * MPU6050 device ORIENTATION
 * Choose which MPU's axis is parallel
 * to your blade axis
 *************************************/
//#define BLADE_X
#define BLADE_Y
//#define BLADE_Z
/************************************/

/*
 * BLADE TYPE
 *
 * RGB LED OR NEOPIXEL users:
 * Comment the following line will
 * disable and remove all LEDSTRINGS
 * blocks from compile
 *************************************/
//#define LEDSTRINGS
#ifndef LEDSTRINGS
#define LUXEON
#endif

#ifdef LUXEON
/*
 * MY_OWN_COLORS
 * If you want to manually specify your own colors
 */
//#define MY_OWN_COLORS
#define FIXED_RANGE_COLORS

static const uint8_t rgbFactor = 100;

# ifdef MY_OWN_COLORS
/* COLORS
 * Number of colors YOU defined in getColor function
 */
#define COLORS		 		3
#else
/* COLORS
 * Number of colors to chose from
 * Range : 6<->600
 * Default: 48
 */
#define COLORS		 		48
#endif
#endif
/************************************/ // BLADE TYPE

/* MAX_BRIGHTNESS
 *
 * Maximum output voltage to apply to LEDS
 * Default = 200 (78,4%) Max=255 Min=0(Off)
 *
 * WARNING ! A too high value may burn
 * your leds. Please make your maths !
 ************************************/
#define MAX_BRIGHTNESS		200

/* LIGHT_EFFECTS
 *
 * Enable / disable  all light and
 * color effects of the blade
 * If you a device with a CPU wich is not
 * an Atmega328 : COMMENT THIS
 ************************************/
#define LIGHT_EFFECTS

#define BLASTER_FLASH_TIME  3
#define CLASH_FLASH_TIME  	1

/* WRIST_MOVEMENTS
 * If you want to enable/disable
 * wrists twists movements
 *************************************/
//#define WRIST_MOVEMENTS

/* DEEP_SLEEP
 * If you want to enable/disable
 * deep sleep capabalities
 * If you a device with a CPU wich is not
 * an Atmega328 : COMMENT THIS
 *************************************/
//#define DEEP_SLEEP
#ifdef DEEP_SLEEP
#define SLEEP_TIMER			300000 //5min = 300000 millisecs
#endif

/*
 * PINS DEFINITION
 */

#ifdef LEDSTRINGS

#define LEDSTRING1 			3
#define LEDSTRING2 			5
#define LEDSTRING3 			6
#define LEDSTRING4 			9
#define LEDSTRING5 			10
#define LEDSTRING6 			11

/*
 * FoCSTRING
 * Enable/disable management of
 * single Flash On Clash ledstring
 *************************************/
//
#define FoCSTRING			13
#endif

#ifdef LUXEON

#define LED_RED 			5
#define LED_GREEN 			10
#define LED_BLUE 			9
#endif

/*
 * ACCENT_LED
 * Enable/disable management of
 * a button accent led
 *
 * If you define ACCENT_LED beware on which
 * kind of pin you defined it :
 * D3,D5,D6,D9,D10,D11 are hardware PWM
 * others must use software PWM
 *
 * RGB LEDS user should choose tu plu their
 * Accent leds on Hardware PWM pin
 *
 * LEDSTRINGS users have no choice :
 * your forced to use Software Accent LED
 *************************************/
#define ACCENT_LED  6  //A2
#ifdef ACCENT_LED
/*
 * Soft or Had PWM for Accent
 */
//#define SOFT_ACCENT
#ifndef SOFT_ACCENT
#define HARD_ACCENT
#endif
#endif //ACCENT_LED

/*
 * MULTICOLOR_ACCENT_LED
 * Enable/disable management of
 * a button accent led
 *************************************/
//#define MULTICOLOR_ACCENT_LED
#ifdef MULTICOLOR_ACCENT_LED
#define RED_ACCENT_LED  16 //A2
#define GREEN_ACCENT_LED  17 //A3
#define BLUE_ACCENT_LED  A7  //.... A7 is input only ...
#endif

#define DFPLAYER_RX			A2
#define DFPLAYER_TX			A3
#define SPK1				A0 //A0
#define SPK2				A1 //A1

#define MAIN_BUTTON			8
#define LOCKUP_BUTTON		4

/*
 * DEBUG PARAMETERS
 */
/* LS_INFO
 * For daily use I recommend you comment LS_INFO
 * When you plug your device to USB uncomment LS_INFO !
 */
#define LS_INFO
#ifndef LS_INFO
//#define LS_DEBUG
#endif

#ifdef LS_DEBUG
//#define LS_BUTTON_DEBUG
#define LS_MOTION_DEBUG
//#define LS_MOTION_HEAVY_DEBUG
//#define LS_RELAUNCH_DEBUG
//#define LS_DEBUG_SLEEP
#endif

#ifdef LS_MOTION_DEBUG
#define LS_SWING_DEBUG
//#define LS_SWING_HEAVY_DEBUG
//#define LS_CLASH_DEBUG
//#define LS_CLASH_HEAVY_DEBUG
#endif

#endif /* CONFIG_H_ */

DJWing79, 2016-03-25:
here's the Lightsaber.ino

/*
 * LightSaberOS V1.0
 *
 * released on: 10 mar 2016
 * author: 		Sebastien CAPOU (neskweek@gmail.com)
 * Source : 	https://github.com/neskweek/LightSaberOS
 * Description:	Operating System for Arduino based LightSaber
 *
 * This work is licensed under the Creative Commons Attribution-NonCommercial-ShareAlike 4.0 International License.
 * To view a copy of this license, visit http://creativecommons.org/licenses/by-nc-sa/4.0/.
 */

#include <Arduino.h>
#include <DFPlayer.h>
#include <I2Cdev.h>
#include <MPU6050_6Axis_MotionApps20.h>
#include <EEPROMex.h>
#include <OneButton.h>
#include <LinkedList.h>
#include <avr/sleep.h>
#include <avr/power.h>

#include "Buttons.h"
#include "Config.h"
#include "ConfigMenu.h"
#include "Light.h"
#include "SoundFont.h"

#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
#include <Wire.h>
#endif

/*
 * DO NOT MODIFY
 * Unless you know what you're doing
 *************************************/
#define CONFIG_VERSION 		"L01"
#define MEMORYBASE 			32
#define SWING_SUPPRESS 		400
#ifdef WRIST_MOVEMENTS
#define	WRIST_SENSIBILITY	1000
#endif
/************************************/

/*
 * BATTERY SAFETY MONITOR
 * Modify to match your baterries specs
 * rule of thumbs for serial wired batteries:
 * LOW_BATTERY = 3200 * Nb_of_battery_cells
 * CRITICAL_BATTERY = 3050 * Nb_of_battery_cells
 *************************************/
#define LOW_BATTERY 		3200
#define CRITICAL_BATTERY 	3050
/************************************/

/*
 * DEFAULT CONFIG PARAMETERS
 * Will be overriden by EEPROM settings
 * once the first save will be done
 *************************************/
#define VOL					13
#define SOUNDFONT 			2
#define	SWING 				800
/************************************/

/***************************************************************************************************
 * Motion detection Variables
 */
[... 1381 lines cut ...]
		uint8_t brightness;
		if (blink == 0) {
			randomBlink = random(7, 15);
			blink++;
			brightness = random(MAX_BRIGHTNESS - 50, MAX_BRIGHTNESS - 70);
		} else if (blink < randomBlink) {
			blink++;
		} else if (blink == randomBlink and randomBlink >= 14) {
			blink = 0;
		} else if (blink == randomBlink and randomBlink < 14) {
			randomBlink += random(7, 15);
			brightness = 0;
		}
#ifdef LEDSTRINGS
		lightFlicker(ledPins, storage.soundStringPreset[storage.soundFont][2],
				brightness);
#endif
#ifdef LUXEON
		lightFlicker(ledPins, currentColor, brightness);
#endif
	}

}

#endif //LIGHT_EFFECTS

DJWing79, 2016-03-25:
I think it's having to do with the interrupt, but I can't be sure.

DJWing79, 2016-03-26:
Ok, I did further testing and yes it definitely is the Interrupt. I can use config mode just fine, the buttons work, the audio works, everything works except when I put it into action mode and it starts reading that interrupt channel. For my setup, I have to have it on pin 3 (interrupt 1), otherwise I will have to spend a lot of money to get the boards and new PCBs made. This is for a project for my wedding (yes my fiance is down with a lightsaber duel, she's a keeper). So any suggestion would be most appreciated.

neskweek, 2016-03-27:
Hello DJWing79 !
You shouldn't have modified that much pins assignement without perfectly knowing their functions.
Some of them are carrying special features(like PWM and interrupts).
A simple solution would be to wire your setup as noted on the schematics.

From what I saw, the Uno as same pin management that the Nano so it should work.

DJWing79, 2016-03-27:
Would the modifications really matter so much? the config mode shows that the colors and audio work properly. and when in run mode, it does the flicker and audio correctly except that it won't turn off or accept any inputs from buttons or accelerometer. Will move the interrupt pin back to pin 2 and test it again (but not move the other pins) and let you know. If that doesn't change anything then I will have to scrap the money i've already spent and make a new PCB and use a nano. if it goes back to working, then maybe you can help me modify the code so it will take interrupt on pin 3, cause if I can't I will still have to scrap what I already have. that's about $100 down the drain if I can't make it work, but oh well...lol.

DJWing79, 2016-03-27:
Ok, I moved the Interrupt back to 0 (pin 2) and now it works again. Is there a modification to the code we can make where we can set it to interrupt 1? It seems like it should work, but not sure where in the code its getting hung up on.

neskweek, 2016-03-27:
DPin2 interrupt is used by I2CDEV and mpU6050 libraries. I don't recommend changing this one.

You can eventually use Dpin3 with the second interrupt, you'll have to move whatever you plug on it to another DPin PWM capable. 

But I don't see why you want to use/change those anyway.

DJWing79, 2016-03-27:
Because I have trinket pro and it can't use pin 2....may have to order nano and new pcb

DJWing79, 2016-04-21:
So back to using pin 3 for interrupt real quick, you said that the IC2DEV and MPU6050 libraries use DPin2, is that hard coded? Or are they looking for an input signal from any interrupt pin? If it HAS to be DPIN2, can the libraries be modified to look at DPIN3 for their interrupt? and if so, where in the code is that? I ask because I have a ton of Pro Trinket boards that I got for this project that are basically paperweights if I can't find some way to make them work. Thanks for the help!

closed 2016-06-13 by neskweek