neskweek/LightSaberOS #4: Eclipse

Soulbp, 2016-03-23:
I'm confused with eclipse, it seems v2.2 is the only one with x64. It does not seem to have the option you mentioned in the arduino section. Also which arduino version is recommended as it complains that mine is newer than the version of eclipse or is this a non-issue?

neskweek, 2016-03-23:
Please read Carefully. From readme.md  :

> WHAT YOU WILL NEED :
>
>Java Runtime Environment 8 Download the right file for your architecture
>Arduino Eclipse __v.2016-03-11_03-29-38__ Again, download the right file for your architecture

[...]

> SET UP YOUR PC:
>
>Download and install Java Runtime Environment 8
>Download the right file for your OS
>Download Arduino Eclipse __v.2016-03-11_03-29-38__

Soulbp, 2016-03-23:
Thank you my original issue was simply not going down far enough on the download page. However unless I'm missing something where is your main loop and setup functions handled? You've done a lot of work here that's for sure. I'm trying to get up to speed with out it all fits together.

Soulbp, 2016-03-23:
Again my mistake it was in LightSaber.ino, why is it named this way instead of .cpp? I ran build all it did not seem to give any errors, hopefully I'll get my sound soon I have everything else. Thank you for your hard work.

neskweek, 2016-03-24:
Hello :)
Thank you 

In Arduino IDE, the main file wears .ino or .pde extension. I just kept that.

Cheers

closed 2016-04-08 by neskweek