neskweek/LightSaberOS #5: Rapid switching between powersave/normal mode

772pilot, 2016-03-26:
When the saber is powered on and left for a random amount of time, the critical battery alert plays and repeats. LS_INFO shows that the saber is alternating between powersave/normal mode continuously.

neskweek, 2016-04-08:
I confirm tht I have the same problem.
Using a PC on windows 10

I don't know which OS your using but I recently discovered that in windows 10 Microsoft added a USB power save management that shut some USB port if it consider it is not used.
It seems it is also present in W7 and W8:
http://www.groovypost.com/howto/usb-selective-suspend-windows-explained/

That might be why,on USB we get critical battery alert.

Please confirm

772pilot, 2016-04-08:
Currently using W7. Disabled selective USB suspend and I'm getting better results. I've run it a couple minutes and so far no issues.

772pilot, 2016-04-14:
Upon further investigation, it's possible that I may have my LOW_BATTERY and CRITICAL_BATTERY constants configured incorrectly. I'm using a 4-cell 7.4V 1600 mAh Li-Ion battery and I'm not sure what my values should be. Where did you get 3200 and 3050 from?

neskweek, 2016-04-14:
As I noted it in the code those parameters are adapted for a 1 cell 4,2V battery.
Search on the net what are the safest low batterys voltage.

neskweek, 2016-06-13:
should be corrected in v1.0.4

closed 2016-06-13 by neskweek