neskweek/LightSaberOS #7: Not able to build in Eclipse

Williamwong086, 2016-04-06:
hi neskweek , Can you help me to look what is the problem , i had follow your instruction step by step but still keep getting error!

11:09:14 **** Incremental Build of configuration Debug_AVaRICE for project LightsaberOS ****
"C:\\Program Files\\eclipseArduino\\arduinoPlugin\\tools\\make\\make" all 
Cannot run program "C:\Program Files\eclipseArduino\arduinoPlugin\tools\make\make": Launching failed

Error: Program "C:\Program Files\eclipseArduino\arduinoPlugin\tools\make\make" not found in PATH
PATH=[C:\Program Files\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/;;C:/Program Files/eclipseArduino/jre/bin/client;C:/Program Files/eclipseArduino/jre/bin;C:/Program Files/eclipseArduino/jre/lib/i386;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Microsoft SQL Server\90\Tools\binn\;C:\Program Files\QuickTime\QTSystem\;C:\Program Files\eclipseArduino]

11:09:14 Build Finished (took 47ms)

neskweek, 2016-04-08:
Hello

Are you using the Eclipse version described in README.md ?

Williamwong086, 2016-04-08:
Hi.. Thanks for reply...Yes...its the link you gave us from the from the README.md...Arduino Eclipse v.2016-03-11_03-29-38

Williamwong086, 2016-04-08:
Btw... I not quite understand the imu calibration part you wrote... Can u guide me how it was done!

closed 2016-06-13 by neskweek