neskweek/LightSaberOS #8: OS locks up when selecting sound font in config mode & clash randomness issues (Arduino nano CH340G clone)

772pilot, 2016-04-21:
When selecting sound fonts in the "sound font" config menu, OS locks up and becomes unresponsive. If the serial connection is open, the saber will reboot. 

Additionally, only two different clashes (first two of six) are played no matter what. This is a recurring issue even after the nano is reset. Happens in both action mode and blaster block mode. All other sounds (ignitions, retractions, swings, lockups, blaster deflections) are randomized.

neskweek, 2016-05-06:
Hello

I can't look at that for the moment but I'll look into it as soon as I get a working device to test it.

In the mean time, can you check that your soundfiles are correctly copied to your SDcard (look at the readme) To be sure of their order, you can copy them one by one after having formating your SDCard,.

772pilot, 2016-05-12:
I reflashed the bootloader to the nano clone and it seems to have resolved these particular issues. The only issue I'm running into now is a very audible gap that occurs when different sounds are played (ex. there's a gap between swings or clashes). Any ideas?

closed 2016-06-13 by neskweek