neskweek/LightSaberOS #10: Wav Sound Problems

Snyperdead, 2016-05-05:
Hello, and congratulation for this work with LSOS !!!

I have a problem, with the sound font aparently.

I'll connected every wire, with the squematic, acording to Andras Kun (aka protonert).

Aparently everything works fine, but when i start the blade, the leds flashing but the sound, stay in static and no hear any more.

i thing the problem are in with the wav sound, because, the mp3 for the config mode are played without a problem.

p.d: sorry for my bad english i hope you understand my redaction without a problem.

Best Regards.

neskweek, 2016-05-06:
Hello 

For the sound the way you did copy it on your card might be to blame. SOund files must be copied in the same exact order they are when you you look at the folder. If they are not it can end up with hasardous behaviour. Look at the readme to see how to do that.

Snyperdead, 2016-05-07:
hi Neskweek

the copy of the files, are realice acording with the documentation, 
format the SD card, 
and drag and drop the files,
for keep the correct order of reproduction.

follow the steps, 2 times assuming i were doing any wrong

closed 2016-07-15 by neskweek