neskweek/LightSaberOS #11: Interrupt Pin

DJWing79, 2016-05-05:
Does the code absolutely have to use Pin 2 (Interrupt 0) or is there a way to have it use Pin 3 (Interrupt 1)? I know you said that other libraries it's dependent on use the interrupt, but are they hardcoding Interrupt 0 or can we set them to Interrupt 1?

neskweek, 2016-05-06:
Hello
Sorry for the late reply.

You can try to, but I don't know how it will react.
I don't remember where it is coded.
I try to find it again before answering but there is no attachInterrupts in I2CDEV nor MPU6050 nor Wire libs.
I think it's located in I2CDEV lib and that it's arduino low level coded.

closed 2016-06-13 by neskweek

DJWing79, 2016-06-17:
If you ever do find a way to move the interrupt from pin 2 to pin 3, let me know. When I do so without changing the code I get the hum sound instantly playing and won't stop. If I change the attachInterrupt from 0 to 1, I get the same thing.