neskweek/LightSaberOS #12: DIYino programming

DJWing79, 2016-05-20:
Have you seen Proto's DIYino and if so, do you know what special steps are needed to program it with LSOS? I just received 6 boards and want to use them if I can.

neskweek, 2016-05-21:
Hello

I did received one DIYino myself and tested it without any problems.
It is fully compatible with LSOS.
Depending of the chips installed on your devices you may (or not) encounter some minor bugs that we didn't discover yet.

DJWing79, 2016-05-21:
So eclipse should recognize it? How did you wire it, I'm a little confused and Proto went on vacation.

neskweek, 2016-05-22:
Eclipse recognize it as an Arduino Nano.

For wiring it's really easy : 
For your LEDs/ledstrng connect the - of your leds to one of the the LS(1 to 6) pin. The + goes to the battery pack
The rest of the connections works exatcly like the schema (buttons / power / speaker)

DJWing79, 2016-05-23:
does the MPU need to be calibrated like in the older instructions? Also, will it actually take LS1 as a designation?

DJWing79, 2016-05-27:
How did you calibrate the MPU portion of the DIYino?

neskweek, 2016-05-27:
Sorry for the late reply.

It calibrates excatly like the Nano/MPU6050 version.
Look at the readme

DJWing79, 2016-05-27:
No worries, just to confirm...wire it up, then have it lay flat in the orientation it will be in, in the final hilt setting and then upload the script to DIYino for the calibration, take down the numbers, plug them into LSOS and then reupload that to DIYino? question...why can't the software read and compare the values for huge changes in acceleration and then vibration for impact? I did that on my first attempts using an ADXL335 and it worked fine, but I had a delay sometimes in the sound and it would sometimes double-font.

closed 2016-05-27 by DJWing79