neskweek/LightSaberOS #13: Main Button Issue

DJWing79, 2016-06-06:
I wired up the sketch on my arduino pro mini and it worked in the mockup, but then I needed to put the main button on pin 3. I got everything wired and it will boot up and the secondary button works, but the main one won't. I went back and checked the wiring and still can't figure it out.

DJWing79, 2016-06-06:
A quick check and it won't recognize main button on any pin. It also does this weird thing when you hold down Aux to enter Config mode, it says "Config Mode: Volume" very quickly and then if try main button nothing happens. I checked the battery and it is fully charged.

DJWing79, 2016-06-06:
Ok, so I found something out. I swapped the main button and aux(lockup) button in the program.

Original setup
Main    3
Aux      4

New setup
Main   4
Aux     3

When I upload the sketch and then run it, it is still showing Aux as being on pin 4, not 3, I think this is the problem. How can I force Eclipse to erase the old sketch on the pro mini and upload the new?

DJWing79, 2016-06-06:
So I've even swapped the physical button that is known to work for the one for Main Button, and no matter what pin I put it on, it won't work. I did find out how to make sure changes are uploaded, so that works. But I'm dead in the water without the ability to turn on the saber. An interesting thing, only Pin 4 seems to work for me. I have finally assigned one of the buttons to pin 7 and the other to pin 4 and it works, but it screws up my PCB that I had made, but oh well....might want to look into that.

neskweek, 2016-06-06:
Hello
I was away this week and could not answer.

I don't own a Arduino Pro mini so I don't know how it is wired or which pin correspond to which one on the Nano.

Just a thought tho : how did you wire your button ? Pin X and (+ or -) ?

DJWing79, 2016-06-06:
Pin is positive then negative to ground. Pin 7 and 4 worked but not 3 or 6, didn't test others

closed 2016-06-13 by neskweek