neskweek/LightSaberOS #15: MPU6050 clash detection works but no clue why...?

Protonerd, 2016-06-20:
Hi neskweek,
Yesterday I've been experimenting again with the clash detection, because it still seems to be not as responsive as it should be. I looked at the code and also read the MPU6050 Manual, all about motion interrupts.

It seems there are only 4 parameters playing a roles:

These settings change the full scale range of the accelero/gyro (not yet in LSOS...). These should change the sensitivity of the saber because an LSB is scaled with full range.
// set the fll scale range of the gyro- and accelerometer respectively
mpu.setFullScaleGyroRange(0); //0: 250deg/s | 1: 500deg/s | 2: 1000deg/s | 3: 2000deg/s
mpu.setFullScaleAccelRange(0); //0: 2g | 1: 4g | 2: 8g | 3: 16g

Then there are the settings for the motion interrupt. The first one sets the threshold, but no idea how...I guess the number means LSB. But it's strange, with a setting of 10, we should hear clash triggering all the time even if we just shudder.
mpu.setMotionDetectionThreshold(10); // 1mg/LSB
mpu.setMotionDetectionDuration(1); // number of consecutive samples above threshold to trigger int

I experimented with following things (using 2 sabers both with DIYino):

set mpu.setFullScaleAccelRange(0) and mpu.setFullScaleGyroRange(0) once to 0 and once to 3 (2 extreme values) -> no change in the clash detection, no tangible effect on other motion detection
play with mpu.setMotionDetectionDuration(1) -> no change in clash detection
Funny thing is, my LED string saber is no (without any change) very responsive, even small taps will trigger clash (the right way!), while my new neopixel saber (using also DIYino) sometimes even fail to detect quite hard hits...I re-calibrated both MPUs and also tried with a sketch what they read, they read and are calibrated both perfectly.

So I feel we still do not fully understand how this motion interrupt feature works...although clash is by far the most important feature of a saber (besides the LEDs).

It is more a fundamental topic, not an immediate bug, but I did not find a better place to report/discuss.

Cheers: Andras

neskweek, 2016-06-20:
In fact it's more trickier than you might think :+1: 

mpu.setIntMotionEnabled(true);
mpu.setMotionDetectionThreshold(10); // 1mg/LSB
mpu.setMotionDetectionDuration(1); // number of consecutive samples above threshold to trigger int

are the parameter involved in this "clash" detection.

The thing is that, on a normal use case this system is meant to detect a movement and generate an interrupt. In this config if the system get a movement superior to the treshold on 1 sample it trigers the interrupt => Clash.
The treshold is "so high" for a set of samples so small here, that only a clash triggers it.

For your Neopixel hilt you might just want to lower the treshold or In this build your MPU is maybe moving to much inside the hilt not allowing it to take a hit :p  (just a supposition ;) ).

Protonerd, 2016-06-23:
:) we think quite similar, my first assumption was also that the DIYino is moving around in the hilt and therefore is not rigid enough for a dependable clash detection. But looking at the inside of the hilt I cannot confirm this...It cannot move, there is simply no space left... I can also lover the threshold, of course with 1 even small movements can trigger a clash, with 5+ no change, but what bugs me is that with the 2g full-scale setting you have 2g/32768=0.06mg/LSB -> a threshold of 10 corresponds to 0.6mg, which is less than a twich of a finger... I simply do not undrstand the scaling of this threshold, but the fact that it also does not change behaviour with the full-scale change means it is somehow indenpenent of the full-scale range and resoultion...but then what does it mean? If I increase it to 1000 (1g=16000), I get no more clash at all!
It is also interesting, that even disabling motion_int will not stop the interrupt to be latched into the status register. I can disable all interrupt sources but clash will still work.

neskweek, 2016-06-23:
So when You just twich the card with your finger it clashes ?

Looking at the photo of your setup on your bucket, maybe cables secures it but it might also react like springs or foam, and absorbing the shock.

for motion int, beware: on previous versions of LSOS I didn't saw there was twice this instruction. Maybe you still have those 2.

I maybe have another solution.
I was saving this for a future email but hey ... if it works better on this build. Keep in mind that Ionly starte"d to experiment that this afternoon. Treshold maybe needs tweaking (augmenting) : 
add/modify those lines : 
```
mpu->setIntZeroMotionEnabled(true);
mpu->setZeroMotionDetectionThreshold(100); // 2mg/LSB
mpu->setZeroMotionDetectionDuration(1); // number of consecutive samples under threshold to 
trigger int```

and replace this : 
`mpuIntStatus > 60 and mpuIntStatus < 70`
by this : 
`mpuIntStatus > 30 and mpuIntStatus < 40`

if it works, I'll explain that in this future mail ^_^,

neskweek, 2016-07-15:
Should be corrected in V1.2.1

closed 2016-07-15 by neskweek