neskweek/LightSaberOS #16: Saber freezes while in action mode

Mawg1, 2016-06-23:
Hi,
I have the latest release from June 20, 2016 and I am still experience some difficulties.  the saber is perfectly happy in standby and config mode.  However, once Action mode is entered the Saber will freeze - this happens randomly between 10 and 60 seconds.  Any thoughts or help would be greatly appreciated! 

Thanks

neskweek, 2016-06-23:
Hello
Sorry to read this ...

With your actual config, have you enabled ACCENT_LED flag ? 
What device are you using (Official Arduino Nano, or clone )?

Mawg1, 2016-06-23:
Hi nesweek,

Once I launch action mode the saber will freeze up very quickly.   I have let it sit on the table and it freezes anywhere between 20 and 60 second.   If I swing it and clash it,  it will freeze up as well (about 30 to 60 seconds).

I'm pretty confident I don't have shorts.  I actually built two separate setups to try and troubleshoot.   One uses and official arduino with the FTDI chip,  the other uses the CH34OG chip.   Both behave the same.   Both setups use RGB leds - right now they are cheap thru hole ones while I wait on my tri-rebel.  They do flicker though.   

Only other thing I can think of is that I am using arduino ide to program.   I could never get eclipse to run properly. 

Thanks for your time! 
Mike

neskweek, 2016-06-24:
Hi 

Did you wire your device according to the plan in the wiki  especially SPK- and SPK+ ?

Mawg1, 2016-06-25:
Hi, 

I did double checked the wiring and it is according to the schematic on everything.  I'm traveling but will try this first thing monday morning.  Thank you again for all the help!

jbkuma, 2016-07-14:
I've had a similar issue with v1.0 and v1.1 the serial keeps outputting values, but the buttons become unresponsive.  I can't tell exactly when it hangs because it continues outputting serial info and sound.

neskweek, 2016-07-15:
Hello,
Please try with latest version (1.2.1).

Also please double check your wirings/soldering for short cuts or deffective solderings.

closed 2016-07-25 by neskweek