neskweek/LightSaberOS #19: LSOS v1.2 can't verify sketch

DJWing79, 2016-07-15:
Using Arduino IDE to compile the newest version and I get the following error

`Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\Damien\Downloads\LightSaberOS\LightSaberOS\LightSaberOS.ino: In function 'void loop()':

LightSaberOS:1099: error: duplicate case value

    case 2: // BLADE MAIN COLOR

    ^

LightSaberOS:1000: error: previously used here

    case 2: // BLADE MAIN COLOR

    ^

LightSaberOS:1118: error: duplicate case value

    case 3: //BLADE CLASH COLOR

    ^

LightSaberOS:1022: error: previously used here

    case 3: //BLADE CLASH COLOR

    ^

LightSaberOS:1149: error: 'struct StoreStruct' has no member named 'soundFontColorPreset'

     storage.soundFontColorPreset[storage.soundFont][0] =

             ^

LightSaberOS:1151: error: 'struct StoreStruct' has no member named 'soundFontColorPreset'

     storage.soundFontColorPreset[storage.soundFont][1] =

             ^

LightSaberOS:1357: error: too few arguments to function 'void lightOff(uint8_t*)'

    lightOff();

             ^

In file included from C:\Users\Damien\Downloads\LightSaberOS\LightSaberOS\LightSaberOS.ino:24:0:

sketch\Light.h:40:6: note: declared here

 void lightOff(uint8_t ledPins[]);

      ^

exit status 1
duplicate case value

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

jbkuma, 2016-07-15:
having the same issue.

neskweek, 2016-07-15:
Use version 1.6.5-R5

jbkuma, 2016-07-15:
I installed to 1.6.5-r5 yesterday and double checked this morning.  Still getting the same error.

neskweek, 2016-07-15:
Yeah, My bad....

I missed an error in RGB blade code.
This should be corrected in V1.2.1 I just uploaded.

jbkuma, 2016-07-15:
It checks out now, thanks!

neskweek, 2016-07-15:
Since I can't test RGB blade code, please tell me if everything is working correctly ;)

DJWing79, 2016-07-15:
So it can upload but now the config mode has three new menus it didn't have before, I can't understand the first two but the last says flicker effect. No matter what I do, I can't change the colors. I am using luxeon and myowncolors with colors set to 48

neskweek, 2016-07-15:
That shouldn't be the case.
Are you sure you commented #define LEDSTRING ?
did you cleared previous files in your directory ?

I can't test fire RGB blade, but I can test sound and menu on RGB code, and I don't reproduce this here.

DJWing79, 2016-07-15:
/facepalm

yup, forgot that part, will test it tonight.

neskweek, 2016-07-15:
Does this work ok now ? :)

if so I'll close this one ;)

DJWing79, 2016-07-16:
So I got the ability to change colors, but only 3 colors. I have FIXED_RANGE_COLORS commented out, but it seems to be stuck on the 3 colors only and won't mix them. Its almost as if the getColor function isn't being called

DJWing79, 2016-07-16:
ok for sheer shits and giggles, I commented MY_OWN_COLORS and uncommented FIXED_RANGE_COLORS and now it works with mixing, weird....it's completely reversed from the previous version.

closed 2016-07-25 by neskweek