neskweek/LightSaberOS #20: LSOS 1.2.2 on Arduino clone unable to connect to DFPlayer

772pilot, 2016-07-15:
Using Arduino IDE v1.6.5-r5. LSOS is unable to start; serial monitor indicates that setup stops at DFPlayer connection and loops repeatedly. A repeated pop from the speaker is audible and the following output shown on the monitor:
```
L01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥L01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥˜01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥˜01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥L01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥L01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥
```

neskweek, 2016-07-15:
It does connect to DFPlayer, it bugs after...
Have you previously customize you soundfont.h ?

772pilot, 2016-07-15:
Soundfont.h is completely unedited.

neskweek, 2016-07-15:
what happen if you augment this delay (at end of setup) ?: 

	dfplayer.setSerial(DFPLAYER_TX, DFPLAYER_RX);
	dfplayer.setVolume(storage.volume);
	delay(200);
	pinMode(SPK1, INPUT);
	pinMode(SPK2, INPUT);

772pilot, 2016-07-15:
Changing the delay interval changes the time between speaker pops.

neskweek, 2016-07-15:
:)
Ok... so it changes nothing ...
Ok let's see if it comes from setup or loop.

Please add : 
Serial.println(F("TestA"));
at end of setup() and then tell me if you see it before the crash on serial.

772pilot, 2016-07-15:
Same output:
'''
L01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥L01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥L01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥L01
Connecting to DFPlayer...
Connected to DFPlayer !
Á²¥
'''

neskweek, 2016-07-15:
ok ...

Refresh my memories please : what's your config ?
Did you plugged dfplayer SPK1 and SPK2 to the arduino ? and if yes where ?

Hopefully there's only 3 simple instructions to incriminate... and I think those are faulty :

pinMode(SPK1, INPUT);
pinMode(SPK2, INPUT);

please comment out and tell the results ;)

772pilot, 2016-07-15:
LED string setup with Arduino nano clone, MPU6050, and DFPlayer mini. No modifications at this time.

SPK1 and SPK2 are connected to pins A0 and A1, respectively.  Commenting the two lines above doesn't seem to change anything.

neskweek, 2016-07-15:
ok ...weird.

Have you changed SPK1 and SPK2 value in Config.h (I changed those since last version to A6 and A7)

the pop comes from dfplayer initialisation. Since the delay augment it's frequency, we pass setVolume and delay ...

try to comment those then : 

	soundFont.setID(storage.soundFont);

	/***** DF PLAYER INITIALISATION  *****/

	//setup finished. Boot ready. We notify !
	dfplayer.playPhysicalTrack(16);

772pilot, 2016-07-15:
Oh, I didn't realize those had been changed. I seem to get the same behavior even if I modify SPK1 and SPK2.  I'm sorry, I'm a little confused. Did you want me to comment the quoted code, or the setVolume and delay functions that come before?

neskweek, 2016-07-15:
the quoted ones ;)

772pilot, 2016-07-15:
That seems to do the trick, however there is no sound.

neskweek, 2016-07-15:
ok .... so uncomment everything ...
except this : 
dfplayer.playPhysicalTrack(16);

772pilot, 2016-07-15:
Doing that causes the loop issue again.

neskweek, 2016-07-15:
damn then comment out this one and comment : 

soundFont.setID(storage.soundFont);

if this stops the problem (you still have no sound) add : 
Serial.print(F("storage.soundFont :"));
Serial.println(storage.soundFont);

and show me what you get on serial please.

772pilot, 2016-07-15:
Ah, I think the issue has been resolved. I removed the 3rd case from Soundfont.h and all is well. I previously had a 3rd soundfont stored in EEPROM, I'm guessing the value doesn't get erased when I reupload a new sketch?

neskweek, 2016-07-15:
That's what I was starting to suspect, that's why I asked you about your previous config, but I omitt to ask you about Soundfont.h.
I'll remove the case 3 since there's no font attached to this case to avoid anymore problems.
Thanks ;)

Tell me what you think about swing detection

772pilot, 2016-07-15:
Very nice! Had to turn down the sensitivity quite a bit, but it's very responsive.

neskweek, 2016-07-15:
Thank you 
I did the changes.
If all is ok I'll close this one ;)

772pilot, 2016-07-15:
Okay!

closed 2016-07-15 by 772pilot