neskweek/LightSaberOS #21: DFPlayer with no sound if compiled under Arduino IDE 1.6.10

Protonerd, 2016-08-09:
Somehow the same sketches which worked all right before fail to play sounds if compiled with the 1.6.10 version, although compilation passes. I checked, not a hardware problem. With 1.6.4 it still works fine,

per1234, 2016-08-09:
The first thing I'd suggest to narrow down the cause is to downgrade to Arduino AVR Boards 1.6.11:

1. Tools > Board > Boards Manager 
2. Click on Arduino AVR Boards
3. Select 1.6.11 from the "Select version" menu.
4. Click "Install"
5. Click "Close"
5. Upload LightSaberOS to your saber and check if the sound works

Arduino AVR Boards 1.6.12 upgraded to a new compiler version so there's a possibility that was the cause of the issue. By downgrading to Arduino AVR Boards 1.6.11 you will go back to the previous compiler version, which is the version included with Arduino IDE 1.6.4. You can easily reinstall 1.6.12 by repeating the steps I listed above except select 1.6.12 instead of 1.6.11.

Protonerd, 2016-08-10:
Hi per1234,
Thanks for the suggestion. I'm not sure if we talk about the same thing, I experienced the problems with the Arduino IDE version 1.6.**10**. So according to what your write the compiler used in this version should be still the same as in the 1.6.4. Or do I misunderstand something?

per1234, 2016-08-10:
Ever since the introduction of Boards Manager in Arduino IDE 1.6.2 there are two different version numbers you have to take into consideration when you encounter a bug that occurs in one IDE installation but not another. The **Ardino IDE** version and the **platform** version. These are two different, unrelated numbers. I'm talking about platform version, you're thinking IDE version. The platform used for LightSaberOS is Arduino AVR Boards. Each Arduino IDE version includes the latest version of Arduino AVR Boards at the time of release. It's theoretically possible for you to install Arduino AVR Boards 1.6.12 on Arduino IDE 1.6.4, though the Arduino developers aren't very good at considering backwards compatibility so this might not work in practice. Arduino IDE 1.6.10 installs with Arduino AVR Boards 1.6.12 so I'm assuming that's the version that you experienced the problem with. As for which Arduino AVR Boards version is running on your Arduino IDE 1.6.4, I'm not sure, you'd have to open **Tools > Board > Boards Manager** to check. To further complicate things, when you install a different version of Arduino AVR Boards in one Arduino IDE version it will usually cause the Arduino AVR Boards version for any other Arduino IDE version to also be updated because the Boards Manager updates are installed to a folder, Arduino15 that all Arduino IDE versions share. However, the location of the Arduino15 folder on Windows changed in Arduino IDE 1.6.6 so if you're using Windows installing a different Arduino AVR Boards version will not affect Arduino IDE 1.6.4. The other exception is if you are running an Arduino IDE version in portable mode(by creating a folder named `portable` inside the installation folder). In portable mode the Boards Manager updates will be installed to the `portable` folder and will only affect that Arduino IDE installation.

The key to making this issue easier to look into is to narrow down the possible differences between the two IDE versions is that could cause LightSaberOS to work in one but not in another. These include:

- LightSaberOS code: If you are using the identical version of LightSaberOS in both version then we know that a change in the LightSaberOS code is not the cause.
- LightSaverOS bundled libraries: If you're using the identical versions of the libraries bundled with LightSaberOS then we know a change in one of those libraries is not the cause.
- Arduino IDE bundled libraries
- Arduino AVR Boards bundled libraries
- Arduino AVR Boards core
- Arduino IDE/arduino-builder
- avr-gcc compiler: The same version of avr-gcc(4.8.1-arduino5) has been included with all versions of Arduino AVR Boards from 1.6.7(included with Arduino IDE 1.6.4) to Arduino AVR Boards 1.6.11(included with Arduino IDE 1.6.9) but in Arduino AVR Boards 1.6.12(included with Arduino IDE 1.6.10) the avr-gcc compiler was updated to 4.9.2-atmel3.5.3-arduino2.

That's a pretty long list of potential causes of the issue and it makes the job of fixing it seem very large but the key is to just systematically eliminate each possibility. If the issue goes away when you use Arduino IDE 1.6.10 with Arduino AVR Boards 1.6.11 then we know that changes in the Arduino IDE did not cause the issue and that it was most likely caused by the new compiler version, changes in the core, or in an Arduino AVR Boards bundled library that LightSaberOS uses, the cause will still not be known but the list of possibilities just got way shorter.

Protonerd, 2016-08-11:
Wow, I learned from you a lot sir! I really feel indebted, not least because your suggestion really solved the issue. This we need to warn users about in the readme.

closed 2016-08-11 by Protonerd

per1234, 2016-08-11:
Glad I was able to help! It's a confusion a lot of people have right now. They think that Arduino IDE 1.6.10 was a bad release when actually there are no new bugs in 1.6.10, the problems are all caused by Arduino AVR Boards 1.6.12, which can affect any IDE version it's installed in.

What was the solution you found?

Protonerd, 2016-08-11:
Like you described I installed the other AVR boards version 1.6.11 and then my sketches worked as before. I.e. I had a general issue with the DFPlayer lib not recognised by the 1.6.12.

I think it is confusing because the versioning scheme is very similar. At least they could have selected a slightly different versioning.

BTW, does this version have to do something with the boards.txt file in the IDE installation? I have an issue with editing this file (just filed a question on the Arduino Forum).

per1234, 2016-08-11:
> I.e. I had a general issue with the DFPlayer lib not recognised by the 1.6.12.

Not recognised or just not working correctly?
> the versioning scheme is very similar.

The reason the versions are close is because before the Boards Manager feature was added there was no separate version so the first version of Arduino AVR Boards released had the same version as the IDE it was split off from. They've drifted apart because Boards Manager allows Arduino to release new versions of Arduino AVR Boards separately from the IDE releases if they fix bugs or add features. It's certainly confusing though.
> BTW, does this version have to do something with the boards.txt file in the IDE installation?

That's more likely to be an IDE issue than an Arduino AVR Boards issue. I've responded to your post on the forum.