neskweek/LightSaberOS #22: A bit confused with OS uploading...

LightOSproblems, 2016-08-31:
Hi neskweek!

Your project is just awesome! I'm building a lightsaber with same electronics, but i have a problem. I keep trying to upload the OS just as in your tutorial, but it just isn't working for me... 
 So I know it might be too much to be asked, but i still ask you: Could you please make a detailed video tutorial of uploading and maybe even building one of those sound boards?

Best wishes

closed 2016-10-11 by LightOSproblems

Protonerd, 2016-10-12:
I can help with the soundboard topic: just order a DIYino Prime, then you have everything you need integrated on a tiny board.
As to the programming, if you have installed the Arduino IDE (refer to the Arduino page for instructions) and you have the right drivers, it must work.

LightOSproblems, 2016-10-20:
Hi!

Thanks for answer. I recently bought a new arduino nano and uploading worked on it :) . Because I have already other parts shipping and I started building the system, I won't buy the DIYino Prime this time.

stay cheeki breeki