neskweek/LightSaberOS #23: Luxeon

Argathena, 2016-09-11:
Have all parts and chassis, down to coding. I do not have required knowledge to program. I know from reading the forum that I need to add "#define LEDSTRING" or do I put "#define LUXEON" 

Also I think I need to set the pins? 

Really I just need to know what to change and where in order for LSOS to work with Luxeon Led.

faeton13, 2016-09-11:
It you wire everything according to scheme in readme catalogue than you only need comment "#define ledstring" and in uncomment "#define luxeon" in config.h

Argathena, 2016-09-11:
Thanks for quick reply. So at the top of the config.h, put "#define LEDSTRING#

Nothing else?

Argathena, 2016-09-11:
#define LEDSTRING

TYPO

faeton13, 2016-09-11:
As i remmember everything works fine for me after that

Argathena, 2016-09-12:
Great! Thank you! One finally question. Do I need to have it fully wired to upload to it? I'm getting errors when uploading and need to narrow it down.  I just want to make sure that I can upload and then wire it.

per1234, 2016-09-19:
>I'm getting errors when uploading

Are you still having problems @Argathena?

Argathena, 2016-10-03:
Uploading is complete.  The Nano had a solder blob that landed on the atmega chip pins. I managed to desolder that blob, and cleaned the pins up. After that, it uploaded. 

Currently waiting on a 470ohm resistor.  Then I need to work on the power for my luxeon tri rebel. I'm sure my answer is in the forums, but if you have any ideas for the power let me know.  (Do I need mosfets is my main question on power)

jbkuma, 2016-10-03:
Yes, you will need MOSFETs.

Argathena, 2016-12-01:
Order DIYino Prime

closed 2016-12-01 by Argathena