neskweek/LightSaberOS #24: Neopixel blade won't turn off

DJWing79, 2016-09-20:
So when I power down, the neopixels still have lit leds on the some of the segments. How can I make sure they power off all the way, even if they suck juice.

Protonerd, 2016-09-22:
They still become power from the pin you use for the data line.
If you connect the - of the pixel stripe to the GND of the board, please make sure to program L (low, ~0V) to the data pin when you want the stripe to switch off, otherwise the data pin will act as supply.
If you connect the - of the stripe through a FET in order to save power, you have to program H (High, ~Vdd) to the data pin, because in this case if you connect L, you back-connect the GND through the data line.

closed 2016-09-22 by Protonerd

DJWing79, 2016-09-22:
I have the Data pin on 13 and the the Negative going through an N-Fet. Where in the code would I need to make a change? Is this something that can be written into LSOS? A toggle of where you're putting the lines or if not, how about a standard of how you want us to wire the neopixels so they work in LSOS?