neskweek/LightSaberOS #25: Settings Correction for Neopixel

DJWing79, 2016-09-24:
In a previous ticket you told me if I wire Negative through a N-Fet I needed to do something...but I am not understanding what you meant. Where in the code would I make the change?

DJWing79, 2016-09-24:
Also, I have the RGBW neopixels, I think it's throwing the code off, any way to change the settings from RGB to RGBW?

DJWing79, 2016-09-24:
Also, I got a strand of generic WS2812 and I can't get it to change colors. Its always red and won't cycle in config mode, but when using other test codes I can get it to work

DJWing79, 2016-09-24:
NM, figured out the RGB setting, needed to specify pin 3 (my NFET) as String 1, but still having issues with RGBW, it won't change colors correctly

Protonerd, 2016-12-14:
LSOS does not support RGBW type neopixels. Due to current consumption considerations this support is alno not planned.

closed 2016-12-14 by Protonerd