neskweek/LightSaberOS #27: Max_BRIGHTNESS and Transistors

Vulkan1, 2016-10-19:
Hi,
first of all I want to thank you so much for this awesome resource, your making a dream of mine come true.
On the business end I want to make sure im correct in doing this and then ask my questions.
Setup wise I have altered the config.h file and made #define LUXEON un commented and commented the other 2. I have left the color stuff as I assume that the "48" colors means 48 different colors picked evenly throughout the spectrum.

I have gotten to the MAX_BRIGHTNESS and for obious reasons its my concern as I don't want to burn any LED's. `/* MAX_BRIGHTNESS
 *
 * Maximum output voltage to apply to LEDS
 * Default = 100 (39,2%) Max=255 Min=0(Off)
 *
 * WARNING ! A too high value may burn
 * your leds. Please make your maths !
 * BE VERY CAREFULL WITH THIS ONE OR 
 * YOU'LL BURN YOUR BLADE'S LED 
 ************************************/
#define MAX_BRIGHTNESS		100
`
I don't know that the 39,2% is. and with 255 being the max, can I assume that is 100% of my batteries power (being 7.4v). If using the luxeon star, my red LED will have a max voltage of around 3.3, therefore I would want about 2.9volts to each pin. this being around 40% of total voltage, ill need MAX_BRIGHTNESS 99. I can then leave MAX_BRIGHTNESS at 100?

last question is with your diagram and the transistors, your diagram has all transistors leading to a board labeled BLADE MALE, and you show only one LED attached. @pin 8 of the BLADE MALE is the voltage input that goes though the LED, and then to the transistor. In my case I would run all the Vin's to the LED's of the luxeon star from pin 8 and then the - of the leds to the transistors of pins 3,5,6 @the arduino. 

If I need to clarify anything please tell me, my explanations can be a but fluffy

Protonerd, 2016-10-25:
Hi Vulkan1,
The PWM value determines the voltage at the PWM capable output, i.e. if you set 100, it is 39.2% of the 5V (or generally speaking the voltage the pads are supplied with). If we stay with 5V, it means 1.95V on the gate of the mosfet transistor you drive your LEDs with. But it's not the same as your voltage over the LEDs!!! The reason is that the channel resistance of a mosfet is not a linear function of the gate voltage and different mosfets have different threshhold voltages and Idd vs. Ugs characteristics. Therefore the topic is more complex than simply taking a value. Therefore the rule of thumb is: select a resistance to limit your current such that even the full battery voltage will not harm your LEDs. Then you can even go up to 255 PWM value without any risk. If you do not want to do that, then you have to characterise your setup (the whole chain from PWM through mosfet to the LEDs)

Vulkan1, 2016-10-25:
Sorry I'm very new to MOSFET's and have used one with failure. What is a pwm value, and can you point me to a website which I can read about this. My batteries are two 3.4v 6000mah batteries in series. This means 7.4v coming off the positive wire. This is the same setup on the diagram from the read me

jbkuma, 2016-10-25:
A search for "PWM" will tell you everything you need. PWM = pulse width
modulation.The purpose of the MOSFET in this instance is to translate the
PWM signal into a higher current.  The value of the PWM value affects how
much time the output is "on", a lower value will reduce the apparent
brightness of your LEDs.  You should always resistor your LEDs so the max
value is safe.

On your batteries.. What type of battery are you using? The highest
capacity 18650s are 3500mah.  6000mah is likely a false number usually
claimed by low quality manufacturers.  If the batteries are low quality
they may not be able to handle the draw necessary to operate your saber
properly.

Vulkan1, 2016-10-25:
http://m.ebay.com.au/itm/2x-UltraFire-18650-3-7V-5000mAh-Li-ion-Rechargeable-Flashlight-Battery-Charger-/172319497566?nav=SEARCH

Those are the batteries.
As what your saying is. I need to put a resistor between (on the circuit diagram) the batteries and LED's to bring the voltage down from 7.4 to 3v

Vulkan1, 2016-10-25:
Or I could use these 2
http://m.ebay.com.au/itm/2000mAh-16340-Rechargeable-Li-ion-Battery-for-LED-Flashlight-CR123A-Charger-/381705942727?nav=SEARCH&varId=650764221434

Vulkan1, 2016-10-25:
Okey so online research has shown the first link are Bullshit batteries. Thanks for notifying about these. I have decided to use Samsung 18650's, two in series to get the desired 7.4v and will use this to charge them indevidually.

https://www.jaycar.com.au/arduino-compatible-lithium-battery-usb-charger-module/p/XC4502

Ignore all my shit above

closed 2016-12-06 by Vulkan1