neskweek/LightSaberOS #29: Sound files corrupted

markturansky, 2016-11-26:
When attempting to extract SDCard.7z on Mac OSX (Sierra) using The Unarchiver, I get the error message "Could not extra the file "..." from the archive "SDCard.7z": The archive file is corrupted"

This happens for all files in the archive.

faeton13, 2016-11-26:
Have you tryed to download it once again?

markturansky, 2016-11-26:
I tried opening the archive via what I got from 'git clone' and I also just attempted to download the archive directly from github. Same error (the archive hasn't changed, so I wouldn't expect that to work).

per1234, 2016-11-26:
Works for me using 7-Zip.

Protonerd, 2016-12-14:
Please, as mentioned by per1234, use 7-zip (free ware, works perfectly!). I close the issue.

closed 2016-12-14 by Protonerd