neskweek/LightSaberOS #30: Luxeon cre LED

Argathena, 2016-12-01:
This is a shot in the dark, but i have a Luxeon Cre Tri Led and as far as i can tell LSOS in its current form is not compatible. The options that i see for luxeon involve an rgb Luxeon. The Luxeon Cre that i have has 2 blue led and 1 white. When the blade isnt "clashing" the two blue lights are On. When a clash happens the blues turn off and the single white tuelrns on. From what i can see with my limited c++ knowledge, this is not how the luxeon portion works. If im wrong please point me in the right direction. If im correct, are you or anyone reading this willing to work and write the code with me. Once again im am very limited in my coding skills. I would also consider paying for someones time to make the code.  I do thank anyone who is reading this.

jbkuma, 2016-12-01:
It doesn't really matter what the three colors are, LSOS will let you set
your blade, flash, and blaster colors without changing the code.  The
config menu will cycle through your dice combinations.  Just set them to
your two blues, white, and white.

Protonerd, 2016-12-01:
What jbkuma wrote is perfectly true, it's already all there. If you want to tweak the Code for different combination of the 2 blue LEDs and the one White, look for this one in Light.cpp:
`void getColor(uint8_t color[], uint8_t colorID) {
  color[3] = colorID;
  switch (colorID) {
    case 0:
//Red
    color[0] = 200;
    color[1] = 0;
    color[2] = 0;
    break;`

[0] is normally red, [1] is green and [2] is blue, but if you have different LEDs, just feel free to adjust the values (at first do not go beyond 200). For instace 0,0,200 would be pure White if you wired your White LED to the 3rd PWM channel. Hope it makes sense.

jbkuma, 2016-12-01:
To be clear, the config menu is the audible menu accessible by a long press
of the auxiliary button while the blade is extinguished.

Argathena, 2016-12-02:
Well its apparent that my knowledge is even less and the research Ive done is lacking. I had no idea that the config menu changed things like that on the fly. Thank you guys for the push in the right direction!

Argathena, 2016-12-05:
Wow this was staring me in the face.  I had the issue of Arduino IDE 1.6.4 not playing sounds, but I did proper research this time and corrected that and i was able to successfully hear the config mode and adjusted my main color and clash color.  It took me a few minutes to tweak the code but i finally managed 2 blue and white, so Thank you!

closed 2016-12-05 by Argathena