neskweek/LightSaberOS #31: Config not saving to eeprom

Argathena, 2016-12-05:
Between playing with "my own colors" and the config menu, I managed to get my saber playing sounds and using the correct leds (2 blues 1 white Luxeon Cre).  My issue is that the volume and main color and clash color is not saving. Every time I remove power, switch to usb, change code, upload, and reconnect my battery, the config is lost and i have to set it all again. Is this working as intended?  I would like to have my settings permanent each time I charge my battery. Thank you again guys, youve been extremley helpful!!

Protonerd, 2016-12-05:
This observation might be true, I recall DJWing had the same problem, but it only happens if you have a single sound font. I don't know the reason - probably something trivial if I had the time to look into it - but if you just add one other sound font, saving the config works. Even if you do not have another sound font, you can simply double up the one you have and modify the Soundfont.h accordingly. Let me know if it fixes the issue. If you feel like debugging, you might as well fix the bug as well :) after all it's open source, find a bug, kill it.

Argathena, 2016-12-05:
I shall give it my best, just wasnt sure if I was doing something wrong. Thank you for the quick reply

closed 2016-12-05 by Argathena