neskweek/LightSaberOS #32: Buzzing sound

Vulkan1, 2016-12-06:
After having uploaded the code the board, when connecting the DFplayer, a buzzing/crackling sound is playing, once connected it stops. It can occur in opperations when using the saber, ive been at it for a while, from when I had no sound, to where I can get most sounds playing. If it helps, the buzzing/crackling/static happens when I power up to volume 21, it can occur, I don't even get the beep, just static, to the point where I think the player crashes and plays no sound.

Another issue I have is no swing sounds, not sure if the accelerometer is even picking them up. I know clash works as that plays.

Hope you can help resolve this. Currenty using LSOS 1.3

jbkuma, 2016-12-06:
Sounds like the DF player is being underpowered. A fully charged 18650 is
usually sufficient, but the DF will start misbehaving sooner than you might
hope.

Vulkan1, 2016-12-06:
My thoughts were just that, I'm running it off my PC, I'm just using a temp speaker that I scavenged, I think it's drawing too much power thus causing the DF to under power. I'll look at switching it out for testing.

Now the second issue why my accelerometer/gyro isn't detected not swings.

jbkuma, 2016-12-06:
Have you run the calibration and set the values?

Vulkan1, 2016-12-06:
yea, and input them to here. 

`/* UNIT1 */
	mpu.setXAccelOffset(-1059);
	mpu.setYAccelOffset(-1912);
	mpu.setZAccelOffset(1525);
	mpu.setXGyroOffset(-217);
	mpu.setYGyroOffset(1);
	mpu.setZGyroOffset(54);`

those being my values, one issue I did have an in order to solve it was having to comment this line, but what I don't set is the clash sensor is working while accelerometer isn't.

` //devStatus = mpu.dmpInitialize_light();  // this is a ligter version of the above`

this came up as undefined, thus not allowing me to upload, commenting allows me to upload it, but the clash still works without it. haven't been able to test it with it uncommented.

(also tested on headphones, no issue, which is perfect)

Protonerd, 2016-12-07:
Did you try to comment out this one?
#define MPUCALOFFSETEEPROM

If this is active as compile directive, the code assumes that the MPU calibration values are stored in EEPROM, this is how I ship out all DIYino devices. But I prepared the code for home-brewed solutions, it's just that you need to be aware that there are 2 methods of storing the calib values.
Alternatively you can use this one:
https://github.com/Protonerd/DIYino/blob/master/MPU6050_calibration.ino
This is my own tweaked MPU calib function, original from Jeff Rowberg, modified.
It will ask at the end if you want to write the calib values into the EEPROM.

Vulkan1, 2016-12-07:
ill give your code a shot and save it to the EEPROM, and ill see if it can pull it from their

Vulkan1, 2016-12-07:
Okey so heres what my LSINFO gives.

`L02
EEPROM LOADED
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful
Initializing DMP...
address: 96
output: -1070

address: 98
output: -1888

address: 100
output: 1509

address: 102
output: -210

address: 104
output: -3

address: 106
output: 55

Enabling DMP...
Enabling interrupt detection (Arduino external interrupt 0)...
DMP ready! Waiting for first interrupt...
Connecting to DFPlayer...
Connected to DFPlayer !
Main button click.
START ACTION
`
Ive done a fresh download of 1.3, changed what I needed to in the config file. 
Clash is working, I hit the breadboard, it makes the clashing sound.
swing is still a no go.
this still comes up as being undefined, I assume its irrelevant though as this would be the manual way to input the offsets.
`devStatus = mpu.dmpInitialize_light();  // this is a ligter version of the above`

Protonerd, 2016-12-08:
If you have a breadboard I assume you cannot make proper swings with it without letting all those interconnects flying around. Therefore I propose to try the following: activate motion triggered blaster deflect by double click in action mode and make smaller but powerful movements with the breadboard. They should trigger blaster block sounds. If that works, then indeed your gestures cannot be interpreted as a swing. Please note that neskweek implemented a quite sophisticated swing detection which only triggers based on:
- acceleration of the movement
- coupled with orientation change i.e. different position in space (quaternion based calculation)

Another thing you might check if the FIREBLADE feature is enabled (for neopixels). If yes, there I intentionally deactivate swings.

Vulkan1, 2016-12-08:
I can confirm fireblade is not on, its commented.
I tried blaster mode, no go. Even taped down any potential loose pins and swung the thing like a mad man, no go and lucky no damage.

I turned on all debugs I thought were relevant.
`Main button click.
START ACTION
CLASH	mpuIntStatus=65
CLASH	mpuIntStatus=65
CLASH	mpuIntStatus=65
CLASH	mpuIntStatus=65
CLASH	mpuIntStatus=65
CLASH	mpuIntStatus=65
`

again swung it like a mad man, still no result.

I decided to use the MPU's example code to see if it was picking up acceleration.
`a/g:	-980	-1266	6156	8828	1135	-116
a/g:	-9006	-2842	32767	9893	-24216	-1110
a/g:	-2022	670	32767	13190	-23629	1028
a/g:	10150	292	-14066	10752	-20889	-863
a/g:	158	826	3966	-120	11553	-863
a/g:	-1570	234	7696	-2878	-3511	-1253
a/g:	1014	7842	32767	-4769	9032	-2060
a/g:	-2856	1906	25724	-5779	13012	-869
a/g:	1628	126	6438	-5568	10241	-1646
a/g:	482	356	18246	-4424	9562	-2117
a/g:	-3028	-188	20884	-2897	5590	-2203
a/g:	-1820	-334	14684	-544	1739	-2069
a/g:	-2596	-644	15504	1155	-2072	-2072
a/g:	-2998	-866	18074	2866	-5120	-1384
a/g:	-2454	-514	16584	3885	-6636	-673
a/g:	-2176	-192	15398	4350	-7836	-128
a/g:	-936	406	17306	4730	-8770	178
`
getting a lovely range of values. if im interpreting this right, it looks like the hard ware is in order.

Any other ideas?

Vulkan1, 2016-12-11:
Still no chance, ideas anyone

Protonerd, 2016-12-11:
Do you have swing sounds? To see if swing is triggered, try to put a Serial.println("swing"); in the code where the swing sould react. See if you can find a swing on the serial monitor.
You can also try to play with the swing sensitorivy parameter, it should be in the Soundfont.h, one per sonud font. Anything below 1000 should be pretty sensitive.

Vulkan1, 2016-12-11:
I'll give it a shot, just finishing moving it from the breadboard to perf board

Protonerd, 2016-12-14:
Any update? Can this issue be closed?

Vulkan1, 2016-12-14:
Sorry I have been soldering it all up and ran into issues. Close it for now seeing as we're working on a new issue and I'll open it in later time once I get the soldering done. I'm looking into making a PCB with some chloride solution. Makes mine a lot cleaner. I'll look into the problem once I've done all that

closed 2016-12-14 by Vulkan1