neskweek/LightSaberOS #33: Nothing happens pressing main switch

bfenty, 2016-12-10:
I've wired a momentary switch for the main switch, and when it connects to ground, nothing happens at all. It grounds pin D12, like it shows in the wiring diagram. I've got this on a breadboard right now. When I connect the power, I get a hiss sound out of the speaker, but that's all. Come to think of it, I don't get anything either when I ground the aux switch. Have I wired something incorrectly?

Vulkan1, 2016-12-10:
What speakers are you using,  I had the same issue with a buzzing when turning on my saber. The speakers were drawing too much power and cause the DFplayer not to function. What I did was get some apple headphones and just temporarily wrapped wire around the connections and connected it to the speaker output. Give that a shot

bfenty, 2016-12-10:
I think my speaker might be a bit too large, but that's not really the issue. I am getting the 'boot up' sound out of the speaker, and using the Serial Monitor on the PC, I get `Connecting to DFPlayer...Connected to DFPlayer!`

But when I press the 'main' button, which connects pin D12 to ground, I do not get any response. There is no sound from the DFPlayer and my LED does not light up. Nothing comes up on the Serial Monitor (I am not sure if it is supposed to, or not?) I cannot think of what I did wrong. I do not have the gyro module attached, but I just wanted to get the sound player working, for now. I thought it would at least come on and play the standard hum sound.

Protonerd, 2016-12-10:
If you do not have the MPU, the sketch will stop in setup seeing that the acc/gyro cannot be initiated. You need that MPU!!! You can disable this check, but you need to do it on your own.

Vulkan1, 2016-12-10:
What LSOS version are you using. And your speaker will be drawing too much power.  I had the same issue I was using a speaker from some disassembled desktop speakers. Try it with some headphones. 

Also scroll down to the bottom of LSOS sketch and uncomment LS_INFO and post the boot info

Vulkan1, 2016-12-10:
Or what Proto said. I was thinking that but wanted to check :p

bfenty, 2016-12-10:
I have connected the MPU now, and still nothing. I will post the boot info as soon as I can figure it out.

Thank you for all of the help, I am pretty new to Arduino so I apologize if my questions aren't very good...

bfenty, 2016-12-11:
Well, it looks like the MPU is failing.

`ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ 
DEFAULT VALUE
Initializing I2C devices...
Testing device connections...
MPU6050 connection failed
Initializing DMP...
DMP Initialization failed (code 1)
Connecting to DFPlayer...
Connected to DFPlayer !
`

What do I need to do about it? I'm pretty sure my connections are solid.

Vulkan1, 2016-12-11:
I would check ur wiring, that you for sure have lsos 1.3. 

You also need to calibrate it, and I would run the example raw for the mpu to see if it is putting out changing values. Otherwise it could be faulty

bfenty, 2016-12-11:
Got it working! I was on lsos 1.0, first of all. there was also a problem with the MPU connection, as you thought. The main button is still a little touchy, doesn't seem to trigger every time. I will keep playing with it, but it appears to be working much better now. Thank you for the help!

bfenty, 2016-12-11:
Meant to post the Serial output for you in the last post.

`L02ÿÿÿÿÿÿÿÿ 
EEPROM LOADED
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful
Initializing DMP...
address: 96
output: -1

address: 98
output: -1

address: 100
output: -1

address: 102
output: -1

address: 104
output: -1

address: 106
output: -1

Enabling DMP...
Enabling interrupt detection (Arduino external interrupt 0)...
DMP ready! Waiting for first interrupt...
Connecting to DFPlayer...
Connected to DFPlayer !
START ACTION
END ACTION
START ACTION
END ACTION
`

bfenty, 2016-12-11:
So this may need a new issue, but...

I have gotten everything working, but now the incorrect sound files play. I read the section on copy order, and was very careful to follow it. Here's what I did:

- formatted the SD card FAT32 (used quick format if it matters)
- drag/dropped all files from the 'soundfont1' folder onto the root of the SD card.
- that didn't work, so I formatted again, then copied each file in the soundfont individually, one at a time, in order. Still had the same problem, the hum sound plays on bootup, and when I press the main button, a blaster hit effect plays instead of the startup sound.

I have tried copying on both a Mac and a PC, both running latest OS.

Protonerd, 2016-12-11:
Care to copy the config sounds first ;o) ? Read the Soundfont.h, you can find the order the code expects on the SD card.

  case 1:
      // soundFont directory 01 is reserved for config menu sounds
      break;
    default:
      // If you specify a ID number not defined here you will end up
      // on the first defined soundfont
    case 2:

bfenty, 2016-12-11:
oh wow. I'm sorry, if it's really that easy then I'm an idiot...

Thanks for being patient with me, this is a great project! I'm really excited to get it working.

Vulkan1, 2016-12-11:
Also make sure u use the 1.3 sound files with the 1.3 code.

bfenty, 2016-12-11:
I'm still getting incorrect sounds, even after formatting and copying the contents of the 1.3 SDCard folder to the sd card root. Below I've pasted screenshots of the directories, which I drag/dropped as described in the Readme:

[image]

[image]

Vulkan1, 2016-12-11:
Are u using the one button code or 2 button. 1.3 or 1.4

bfenty, 2016-12-11:
1.3

Protonerd, 2016-12-12:
The topic is that the Soundfont.h reflect my working setup with 5 distinct sound fonts, but the SDCard.7z has only a single one, and less files.
As this is a recurring difficulty for users, but I do not want to maintain different combos, I'm going to copy files to that the zip matches my setup. Of course many files will have multiple copies, i.e. all sound fonts will be the same.

Protonerd, 2016-12-12:
Updated SDCard.7z for LSOS 1.4
You can use v1.4, same as 1.3, with some improvements and bug fixes as well as the selectable SINGLEBUTTON feature. If disabled, same as 1.3.

bfenty, 2016-12-12:
Thank you, I will test this out when I get home from work tonight! I really appreciate all of your help!

bfenty, 2016-12-12:
Have you heard of the drivesort utility? It is designed to update the copy order to match the alphabetical order for FAT32 drives. I think LSOS users might benefit from using it.

[http://www.anerty.net/software/file/DriveSort/?lang=en](url)

Protonerd, 2016-12-14:
bfenty, did you try the new SDcard.7zip? Let me know what you get.

wrt to the sort utility, it is something to consider, thanks for linking it in,

bfenty, 2016-12-15:
Yes, the new sdcard.zip worked well. Thank you!

I am working on modifying the code to support toggle switches as well as momentary switches. I've got it updated but not sure how I can contribute it back to the project? Or if you even want that. My setup has a toggle switch is why, thought others might benefit. I just made long press start start the saber and long press stop shut it down.

bfenty, 2016-12-15:
V 1.4 would not compile for me, but I used the soundfont.h from 1.4 with the rest from 1.3 and it worked.

Protonerd, 2016-12-18:
I discovered that I forget to update the MPU lib in 1.4... sorry for that! Now it's fixed.

As to contribution, glady, but the thing is I''m also not that experienced in using GitHub, that is one reason I still keep using branches instead of updating the master... so in theory I welcome each and every one of you willing to contribute, it;s just that I need to learn first how this can be done in a professional fashion. Any suggestions are welcome.

bfenty, 2016-12-18:
I am not the person to ask for help with github. I will keep working on my own a little bit and if I can figure out how, I will be happy to contribute further. I really like the idea of LSOS and appreciate you putting it together.

I'm hoping to 3D print a 'capsule' that will hold the different parts together, make wiring easier, and help avoid solder breaking apart. Once I have done that, I will post a link so hopefully others can use it.

This thread has gone away from the main topic, I'm not sure if we should close it now? Or if there is a better way to discuss the project ongoing? Again, I am not very familiar with Github.

Vulkan1, 2017-03-25:
Yes this can be closed, thought I might look at the issues and see what's resolved.

misrahim, 2017-06-06:
One more question, where do i get the LSOS v 1.4?
I dowload the last one and does nothing for me