neskweek/LightSaberOS #34: Wrong soundfile order?

LightOSproblems, 2016-12-12:
Hello!
 I uploaded the OS, every function works fine (long-short presses etc.), but the sounds are in wrong order. (Instead of "Boot" it plays "Flicker effect" etc.). I did the DRAG AND DROP NOT COPY AND PASTE! I have tried everything: changing the order of sound files, different formatting types of SD card, even changed the numbers and names, but can't get it fixed.

Anyone can help?
Thanks

jbkuma, 2016-12-12:
Make sure you have your counts correct for each sound in the font.  I spent
a couple hours trying to figure out what the heck I was doing wrong before
I realized I had accidentally left it configured for 3 ignition sounds for
my font instead of only the 1 that I loaded on the SD card.

Protonerd, 2016-12-14:
try the new SDCard.7zip I updated for LSOS 1.4.

closed 2016-12-14 by Protonerd