neskweek/LightSaberOS #35: Error Compiling?

aryaredha, 2016-12-14:
hello?
I had a problem with arduino LighsaberOS that there are "compiling error", how do I solve the problem

Thanks

Protonerd, 2016-12-14:
care to share the error log?

aryaredha, 2016-12-15:
this is the problem that I found and I can not upload to my arduino nano
[image]

per1234, 2016-12-15:
You need to install the libraries. Please see step 5 of https://github.com/neskweek/LightSaberOS#set-up-for-arduino-ide.

By chance I was able to decipher the problem from the screenshot this time but you need to understand this is not the proper way to ask for help. You have only showed a screen shot of a fragment of the error. The full console output won't fit in the console window as you have it so you would need to scroll up to see the full output. When you need to share an error you should click the "Copy error messages" button on the right side of the orange bar and then paste the error as text **USING CODE TAGS**(<> button on the toolbar).

aryaredha, 2016-12-16:
I am sorry before, but I still found a problem I follow the instruction Move all folders inside LightSaberOS/Libraries to {sketchbook folder}/libraries. its shown like this 

Arduino: 1.6.7 (Windows 10), Board: "Arduino Nano, ATmega328"

Build options changed, rebuilding all
D:\diy lightsaber\Lightsaber One Button\LightSaberOS\LightSaberOS.ino: In function 'void setup()':

LightSaberOS:267: error: 'class MPU6050' has no member named 'dmpInitialize_light'

   devStatus = mpu.dmpInitialize_light();  // this is a ligter version of the above

                   ^

exit status 1
'class MPU6050' has no member named 'dmpInitialize_light'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

help me please

Protonerd, 2016-12-18:
Oh, this seems to be a recurrent issue. The _light function is new and you must download the new MPU library to be able to use it. But it seems it not very straightforward. I guess the problem can be if you
- either did not download the new MPU library, in which case the compiler will not find it
- or you did download it, but since your libraries might be references from a source other than the new one, still the old lib is in use.

Anyway, try to download the new lib, purge your PC of all old MPU libs.
But if all else fails, just remove the _light from the function all. It will then call the bigger function, which does the same, but takes a lot more code space.

aryaredha, 2016-12-19:
I've tried to remove _light a few days ago and it worked
I try to uncomment devStatus = mpu.dmpInitialize_light (); but there are still problems. as you say just remove _light and it bekerta. so far I have not found the problem after remove it

I use arduino nano, mpu6050, dfplayer, and led single

Thank you for the advice.

Protonerd, 2016-12-19:
OK, I found the problem. In 1.4, the libs were copied from the master branch, so the _light function was not part of the cpp file. Now I fixed that. Please do a full download of 1.4 again and let me know if it works now.

aryaredha, 2016-12-21:
hello protonerd , I'm sorry for the late I wanted to let you know that LightSaberOS v1.4 its worked with a copy of the library in the master branch.

shiryu55, 2017-02-09:
Protonerd , Copied libraries of the new version, but does not compile

Arduino: 1.6.0 (Windows 7), Board: "Arduino Nano, ATmega328"

LightSaberOS.ino: In function 'void setup()':
LightSaberOS.ino:229:18: error: 'loadConfig' was not declared in this scope
LightSaberOS.ino:250:12: error: 'saveConfig' was not declared in this scope
LightSaberOS.ino:480:16: error: 'InitDFPlayer' was not declared in this scope
LightSaberOS.ino:485:22: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino: In function 'void loop()':
LightSaberOS.ino:525:21: error: 'HumRelaunch' was not declared in this scope
LightSaberOS.ino:545:43: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:565:17: error: 'motionEngine' was not declared in this scope
LightSaberOS.ino:578:16: error: 'motionEngine' was not declared in this scope
LightSaberOS.ino:598:47: error: 'LoopPlay_Sound' was not declared in this scope
LightSaberOS.ino:612:42: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:653:53: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:882:48: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:907:21: error: 'HumRelaunch' was not declared in this scope
LightSaberOS.ino:935:22: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:953:22: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:960:22: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:977:41: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:1165:44: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:1191:15: error: 'saveConfig' was not declared in this scope
LightSaberOS.ino:1203:22: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino:1241:26: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.ino: In function 'void HumRelaunch()':
LightSaberOS.ino:1387:38: error: 'LoopPlay_Sound' was not declared in this scope
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

jbkuma, 2017-02-09:
Is there a reason you are still using the 1.6.0 IDE? The code is tested to
work up to 1.6.11.  There are some issues with 1.6.12.  I haven't tried
with 1.6.13 or 1.8.x.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

shiryu55, 2017-02-10:
Thank you very much for the constestation, I worked perfectly.

Right now I have a look at your online store, sure there is something interesting that can serve my project.

Protonerd, 2017-02-10:
Is this issue solved now? Can it be closed?

shiryu55, 2017-02-10:
To what I refer to compile and burn the program worked perfectly using version 1.6.1.1

I am a sea of ​​doubts when it comes to properly feed the circuit, and that neopixel strip model use to maximize the brightness of the project. I guess it will be through the test failure method.

The language in English and the technical complications are a barrier to take into account, in Spain there is not much affition to these topics.

shiryu55, 2017-02-10:
For if you want to intruducir in the base of sounds cuts of the voice of vader in Spanish, Spain.

https://mega.nz/#!Mh8nlBrR!Xash8dYgHnz0iC6NrFRKKe51i4bkA9mVKoSwrvzorI0

Vulkan1, 2017-03-25:
I think it's been resolved, and can be closed.

shiryu55, 2017-03-25:
Ok

closed 2017-03-26 by per1234