neskweek/LightSaberOS #37: Trouble with lsos 1.4 or the diyino?

lewisdw1, 2016-12-24:
Looking for some assistance in troubleshooting my diyino and LSOS.  I have a single LED luxeon wired to LS1, using a 3.7V lion battery with 5vdc regulator to the 5v pin, two buttons to D4 and D12, 4ohm speaker on spk1 and spk2.  I compiled and loaded the LSOS 1.4 configured for two buttons.  When I connect the power I see the lights on the board, hear a startup sound for second or so, and the saber light starts to pulse.  It continues to pulse the light, but pressing the buttons does nothing.  I've checked my soldering job several times, the buttons are working correctly according to my multimeter.  Don't know if I have a software or hardware issue.

Protonerd, 2016-12-24:
I'm not sure the download really worked. What you described is the DIYino testsketch I use to test the boards which get shipped with that code. Upon uploading, did you see the red and green leds flash during upload? Did the Arduino IDE report "Done uploading."?

lewisdw1, 2016-12-24:
Yes, I saw the green and red leds, and here's the output of the IDE:
```
Continuing to use /dev/cu.usbserial-AH03KSZY
Ending reset

Launching /Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/tools/arduino/avrdude/6.3.0-arduino6/bin/avrdude -C/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/tools/arduino/avrdude/6.3.0-arduino6/etc/avrdude.conf -patmega328p -carduino -P/dev/cu.usbserial-AH03KSZY -b57600 -D -Uflash:w:/Users/wayne/Documents/workspace/LightSaberOS/Debug_AVaRICE/LightSaberOS.hex:i 
Output:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "/Users/wayne/Documents/workspace/LightSaberOS/Debug_AVaRICE/LightSaberOS.hex"
avrdude: can't open input file /Users/wayne/Documents/workspace/LightSaberOS/Debug_AVaRICE/LightSaberOS.hex: No such file or directory
avrdude: read from file '/Users/wayne/Documents/workspace/LightSaberOS/Debug_AVaRICE/LightSaberOS.hex' failed

avrdude done.  Thank you.

avrdude finished
upload done
```

lewisdw1, 2016-12-24:
You're correct, it's not building correctly.  I'm getting an error about dfplayer.h:
Description	Resource	Path	Location	Type
fatal error: ../../../../Documents/workspace/LightSaberOS/libraries/DFPlayer/DFPlayerSerial.h: No such file or directory	DFPlayer.h	/LightSaberOS/libraries/DFPlayer	line 15	C/C++ Problem

lewisdw1, 2016-12-24:
I'm using Eclipse (sloeber, on OS X), and when I right click, choose import, and select the option to import the libraries, per your instructions, it doesn't show the libraries that are in the lightsaberos-master:
[image]

lewisdw1, 2016-12-24:
Ok, I'm closer!  I got it to compile properly and uploaded.  The on/off button works, the blade lights, but no sound.  I'm on a Mac, so not sure if the file order is an issue.  I formatted the sdcard as fat32, and copied over the ConfigSounds, Saber_JukeBox, and Soundfont1 folders to the sdcard, in that order.  I wasn't sure if you're supposed to the copy the folder structure (the folder and its contents), or just the files from each folder.  I was getting the test sounds before, so I assume the speaker works.  Now, when I apply power, the leds on the board come on, I hear a click sounds, and I can press the button and the blade comes on, just no sounds.

lewisdw1, 2016-12-24:
Still no sound.  I connected a pair of earbuds to make sure it's not the speaker.  You can hear a sound when the blade is lit, but it's the "feedback" from the light, I think, kind a highpitched hum.  I reformated the sdcard on a windows 7 system and dragged the contents over as your writeup mentions, no change.  I'm using 1.3 right now, was 1.4, no change.

This is the output of the serial monitor:
L02��������
Connecting to DFPlayer...
Connected to DFPlayer !

jbkuma, 2016-12-25:
Can you manually play tracks on the DF player? You can do this by shorting
either the next or previous pin to ground. These are the pins opposite the
speaker pins. A long press on the next pin will raise the volume.

Protonerd, 2016-12-25:
Come tovthink about it, I recall that with the Arduino IDE the dfplayer does not work if compiled with arduino avr board definition 1.6.12 or greater. Please look at your board manager and install 1.6.11

Protonerd, 2016-12-25:
Board definition not to be mixed up with IDE version. They use similar versioning but are completely independent stuff.

lewisdw1, 2016-12-27:
I did get it working, wrong libraries as you mentioned.  The "problem" I have now is when I retract the blade, the humming noise continues for another 30-60 seconds after the blade has retracted (led goes out).  Is this normal, or a programming bug?

lewisdw1, 2016-12-27:
I just caught what you said about Arduino 1.6.12 or greater.  I'm running 1.6.16, and can't get 1.4 to compile at all (and see where some others are having issues also).  I'm going to downgrade to 1.6.11 this evening and try it again.  I'm hoping this will also fix the issue I'm having with the soundfonts.  I have the 1.4 soundfonts on my sdcard (all of them), and can't switch between them in the config menu, I think it's only playing soundfont 1 (even though I modified the soundfont.h to have the same config as the 1.4 version so all should be seen.  I'm on a Mac (os x), and I don't know if the file order issue is part of os x like on windows, but I formatted the sdcard, and did a drag/drop copy of the folders from the zip file so they "should" be in the correct order.

jbkuma, 2016-12-27:
I'm using 1.6.13 on win10 with no issues with compiling the latest 1.4 with
#define SINGLEBUTTON commented out.

As far as sound fonts.. have you updated "#define SOUNDFONT_QUANTITY"?
I've driven myself batty forgetting to update that number.

lewisdw1, 2016-12-27:
Yes, 1.3 soundfont.h with these definitions:

#define SOUNDFONT_QUANTITY 5 
#define NR_CONFIGFOLDERFILES 14 
#define NR_JUKEBOXSONGS 5 
#define NR_SF1_FILES 30 
#define NR_SF2_FILES 30 
#define NR_SF3_FILES 30 
#define NR_SF4_FILES 30 
#define NR_SF5_FILES 30 

I'll try to compile 1.4 again this evening and send you the errors I'm receiving.

lewisdw1, 2016-12-28:
Here's the output of the Arduino compiler when I compile:

```
In file included from /Users/wayne/Downloads/LightSaberOS/LightSaberOS.ino:31:0:
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h: In instantiation of 'LinkedList<T>::LinkedList() [with T = unsigned int]':
sketch/SoundFont.h:30:15:   required from here
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:101:6: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]
  root=false;
      ^
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:102:6: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]
  last=false;
      ^
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h: In instantiation of 'LinkedList<T>::~LinkedList() [with T = unsigned int]':
sketch/SoundFont.h:30:15:   required from here
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:121:7: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]
  last = false;
       ^
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h: In instantiation of 'bool LinkedList<T>::add(T) [with T = unsigned int]':
sketch/SoundFont.h:276:18:   required from here
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:192:12: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]
  tmp->next = false;
            ^
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h: In instantiation of 'T LinkedList<T>::pop() [with T = unsigned int]':
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:279:14:   required from 'T LinkedList<T>::shift() [with T = unsigned int]'
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:320:9:   required from 'void LinkedList<T>::clear() [with T = unsigned int]'
sketch/SoundFont.h:150:22:   required from here
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:248:13: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]
   tmp->next = false;
             ^
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:256:8: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]
   root = false;
        ^
/Users/wayne/Documents/Arduino/libraries/LinkedList/LinkedList.h:257:8: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]
   last = false;
        ^
```
Could that be the reason I can't changed soundfonts in the config menu?

lewisdw1, 2016-12-28:
Got rid of the compiler warnings, using the latest linkedlist.h and a few tweaks.  Also using the 1.6.11 version of the boards manager now, sounds are more correct now and the humming sound stops when the blade retracts.  Still not able to switch between the different soundfonts.  I think it has something to do with the way os x is copying the files to the sdcard (and the extra file info that os x writes to the directories).  I'll have to find a Windows machine and try that.  Any other os x users out there that has gotten this to work correctly?

lewisdw1, 2016-12-29:
Lol! I feel a little silly now.  I've been banging my head against the desk trying to figure out why I couldn't switch between the soundfont folders using the config menu, and the file included with the 1.4 version is supposed to have 5 different soundfont folders.  Well, all 5 folders have the same sound files!  So they all sound identical.  Anyone have any other soundfonts they would care to share?

I also modified the buttons source so I can activate jukebox mode in the two button version.

Protonerd, 2016-12-29:
;) Congratulation, challenge successfully mastered. It was a trap I purposefully built in. Anyone bypassing it is a pavan worthy of a lightsaber

jbkuma, 2016-12-29:
You can look on the fx-sabers.com forum for fonts folks have made for free.
You'll have to make some adjustments to the files.  Someone just released a
really nice and feature rich Force Unleaded font.

Saberfont.com has a great repository of fonts for purchase.

Otherwise, sharing fonts others have worked hard to produce is very much
frowned upon.

lewisdw1, 2016-12-29:
jbkuma - Thanks for the links.  I'll try those.