neskweek/LightSaberOS #38: Can't umcomment "#define SINGLEBUTTON" in LSOS 1.4 I want to use aux button.

Sligs78, 2016-12-25:
The default setting when leaving "#define SINGLEBUTTON" causes LSOS to be single button mode. When uncommenting, it gives a compile error at "Buttons.cpp" at line 260 "void mainLongPress() {". 

It gives the error message:
-----------------------------------------------------------
exit status 1
a function-definition is not allowed here before '{' token
-----------------------------------------------------------

=======================================================

I'm trying to use the aux button function because it has the lock-up sound bug fixed.

Also, how do you go down in the menu in LSOS 1.4? I tried to turn the volume down, but the only option was to go louder. I had to change it back to LSOS 1.3 to turn it down.

Protonerd, 2016-12-26:
Please try again now, I fixed the problem.

Sligs78, 2016-12-27:
The aux button is working now, but now there's other problems. When I hold the main button to turn it off, the DIYino resets. It does the same thing when I'm in the config menu and get to the clash color--the power resets before the clash color option shows up.

I also tried that light ignition speed suggestion in the other topic, but it only turned on the first LED. Same with the other numbers.

Protonerd, 2016-12-27:
I recall you mentioned that you use 144 leds. That is too many! Unfortunately the nepixels use up a lot of dynamic RAM, and if the avr runs out of it, it resets. What you describe fits this particular error behaviour. I tried the code up to 120 leds, I do not guarantee nor recommend to use more due to unpredictable behaviour. Even if you have the full 1m of stipe, try to first reduce the count in the code to let's say 100 and try. It will work. Then you can experiment what is the highest number which will still maintain function integrity,

Sligs78, 2016-12-27:
I just tried 100 LEDs and it did the same thing.

closed 2017-01-14 by Sligs78