neskweek/LightSaberOS #39: How do you change the speed of the ignition animation?

Sligs78, 2016-12-25:
I set my neopixel count to 144. I want it to light up faster to match up with the ignition sound.

Protonerd, 2016-12-26:
In line 948 of Light.cpp, you see this:
`for (uint8_t  i=0; i<NUMPIXELS; i++) { // turn on/off one LED at a time`

If you want to make the ignition faster, the only way is to switch on/off multiple leds at a time, like this (5 pixels per cycle):
`for (uint8_t  i=0; i<NUMPIXELS; (i=i+5)) { // turn on/off one LED at a time`

closed 2016-12-26 by Protonerd

Sligs78, 2016-12-27:
I just tried it; it only turned on one LED. I also tried i+2 and it only turned on one.

jbkuma, 2016-12-30:
You can try this code to replace the RampNeoPixel function

void RampNeoPixel(uint16_t RampDuration, bool DirectionUpDown) {
  unsigned long ignitionStart = millis();
  cRGB value;
#ifdef FIREBLADE
  for (unsigned int i=0; i<NUMPIXELS; (i=i+5)) { // turn on/off one LED at a time
     FireBlade();
     for(unsigned int j=0; j<NUMPIXELS; j++ ) { // fill up string with data
      if ((DirectionUpDown and j<=i) or (!DirectionUpDown and j<=NUMPIXELS-1-i)){
        }
        else if ((DirectionUpDown and j>i) or (!DirectionUpDown and j>NUMPIXELS-1-i)){
          value.r=0;
          value.g=0;
          value.b=0;  
          //heat[j]=0;
          pixels.set_crgb_at(j, value); // Set value at LED found at index j
        }      
      }
      pixels.sync(); // Sends the data to the LEDs
    }    
#else
  for (unsigned int i = 0; i < NUMPIXELS; i = NUMPIXELS*(millis()-ignitionStart)/RampDuration) { // turn on/off one LED at a time
     for(uint8_t  j=0; j<NUMPIXELS; j++ ) { // fill up string with data
      if ((DirectionUpDown and j<=i) or (!DirectionUpDown and j<=NUMPIXELS-1-i)){
        value.r = MAX_BRIGHTNESS * currentColor.r / rgbFactor;
        value.g = MAX_BRIGHTNESS * currentColor.g / rgbFactor;
        value.b = MAX_BRIGHTNESS * currentColor.b / rgbFactor;
        }
        else if ((DirectionUpDown and j>i) or (!DirectionUpDown and j>NUMPIXELS-1-i)){
        value.r=0;
        value.g=0;
        value.b=0;      
        }      
        pixels.set_crgb_at(j, value);
      }
     pixels.sync(); // Sends the data to the LEDs
     delay(RampDuration/NUMPIXELS);
    }
#endif
}