neskweek/LightSaberOS #41: Sound Fonts issue

Vulkan1, 2017-02-11:
So ive been working though troubleshoots for a while now. Im now at the point where ive even done my own PCB cause my old board drove me nuts. So far all is good, all connections are perfect and things are where they are meant to be. 
I have an issue, I believe with, and most likely due to my accord, coding.
So im experiencing crackling, buzzing and incorrect sounds. I have all 5 sound fonts on. and checked their number count to the count on the soundconfig. I just cant figure it out. I looked at hardware and to my knowledge of software. 
as lea has said in the past, with some slight rewording.
Help me community, your my only hope.

on a more serious note, if I need to attach anything give me a buzz.

jbkuma, 2017-02-11:
As for the buzzing.. Are you powering the sound off the Arduino? The
dfplayer should be powered directly.  If you aren't using a battery yet, it
could be a noisy power supply.

If you are getting incorrect sounds thay is most likely due to either a
Soundfont.h issue or SD card issue.  Format your card using the tool here
https://www.sdcard.org/downloads/formatter_4/

Are you running a NeoPixel blade? You may be running out of memory. If the
IDE is trying you that you are using more than around 60℅ of RAM, the pixel
blade can cause some issues. Try reducing to 3 or 4 fonts. To test, you can
simply comment out the last case in Soundfont.h and change the number of
fonts without changing the sd card.

Vulkan1, 2017-02-11:
As for the bussing, the dfplayer Vin is connected to the 5v off the arduino. I am using apple headphones to test the sounds.
Blade type is luxeon.
And i will try out the SD formatting. That most likely will be the issue as I forgot the all important step of formatting before moving the files over, various times.

jbkuma, 2017-02-11:
Take the dfplayer off the 5v and connect it to the battery +, that is
almost certainly the cause of your buzzing.  If you are powering via USB
right now you can either power it separately or a smoothing capacitor might
help (I haven't tried this).

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Vulkan1, 2017-02-12:
okey so though the correct formatting, I have arrived to the correct sounds. now to deal with the crackling (more like static), could it be that due to my PCB (it was an iron on technique) copper pads being somewhat crap, that its a bad connection? I have powered the DFplayer off the arduinos 5v before with no issues. Even when supplying the board with an apple ipad charger power supply it had more then it needs with 5v and 2amp's across the board.

Vulkan1, 2017-02-12:
okey so update, for the sake of ease I added an audio jack with quick connects, and pluged the headphones in via that. sound is perfect now. before I used my fingers to hold the wires in place, stupid me didn't think initially that the sweat on my fingers would cause the static, let alone my skin. ive updated my previous comment with the only issue im having.

Vulkan1, 2017-02-13:
When I put heavy motion debug on, this is what im receiving.
`Q		w=nan		x=nan		y=nan		z=nan`

Vulkan1, 2017-02-13:
This is become a working Log XD, so I turned off all but LS Info so I could do the DMPinitalise() without light, knowing it will take more space. Swing detection now works, but I take up 98% of code space now. Any ideas why I was getting the DMPInitiliaze_light() issues?

Protonerd, 2017-02-13:
The _light version works as expected, I do not have any issues with swings. Maybe you have to run the calibration again? BTW, the only known problem with swing is when you have neopixels and want to run the Fireblade FX. IN that case I had to switch off swings, because that branch keeps creating phantom swings, due to a random() command. Root cause is in debug but not understood at all...

Vulkan1, 2017-02-13:
sorry let me resummerlise, 
` devStatus = mpu.dmpInitialize_light(); // this is a ligter version of the above `
gives me the error of  `class MPU6050' has no member named 'dmpInitialize_light' ` when compiling.
I played around and changed it to `devStatus = mpu.dmpInitalise();` and now it works, but it takes up 98% of the drive space rather then 89%. Swings and all work now. in fact ive almost got a working sabre, I need to finish the power supply and do the LED's test before I get the luxeon. 

I wanted to know why I get that error with the "light" version of DMPInitalising.

For all the help though, I wanted to know if I could give back. ive been working on the ultrasabers free sound fonts, adding all the 2min hums to the ends making it perfect. I wanted to know if I could share that, so that people get a stock 5 different sound fonts. Am I allowed to do this or would it breach some right?

Protonerd, 2017-02-13:
You get the error because you either did not download the newest library version for the MPU or you did, but there is a copy of the old library still somewhere on your PC and the arduino ide finds it first.

About the sound fonts, I appreciate your offer but please check with US first. I do not think they meant to share these sound fonts for free. If it came preinstalled on a board, you are not allowed to distribute them.

Vulkan1, 2017-02-13:
That could make sense, I will dig to find this file. With IDE is it possible to bring up all the library's it uses or will I need to find it.

Also here's the page with all the sound fonts, if being their site. They are LSU files. I converted them to Wav's and then edited them for it to work with the LSOS. By the sounds of it, I can distribute them to you all for FREE, with no price. But see what you think.
http://www.ultrasabers.com/Sound-Font-p/soundfont-manual.htm

jbkuma, 2017-02-13:
If you go in preferences and turn on "verbose output" you may see something
like this:

Multiple libraries were found for "Wire.h"
 Used: C:\[...]\packages\arduino\hardware\avr\1.6.17\libraries\Wire
 Not used: D:\[...]\Documents\Arduino\libraries\Wire

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Vulkan1, 2017-02-15:
Worked perfectly, all fixed now.
About those sound files, do you think I can make a compleation of them. Otherwise I'll pull them off the movies.

jbkuma, 2017-02-15:
Glad you got it working.  We wouldn't be able to officially share any font
without express permission from the creator and/or copyright holder.  If
you get permission from the creator you might share it on one of the FX
forum.  People always appreciate a good font that someone is willing to
share.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

closed 2017-02-18 by Vulkan1