neskweek/LightSaberOS #42: Lightsaber os 1.4  imu orientation , Blade effects, dfplayer fed

shiryu55, 2017-02-12:
1) Determine IMU orientation,  

[url=http://subefotos.com/ver/?d140df178fe03f9871018cbc53592d14o.jpg][img]http://thumbs.subefotos.com/d140df178fe03f9871018cbc53592d14o.jpg[/img][/url]

In this position taking into account that the usb will be in the bottom of the handle of the sword which position is

2) speaker + --to -- a1,,, speaker -to--  a0

Get effect, connections would be these in the lightsaber os 1.4 with arduino nano v3.

In case of being, when increase of sound, in other tests that realize with the lightsaber os master, with a led believes, when increase the volume increases the effect. You can get a less noticeable effect when you still sound.

3. Direct feeding 5v from arduino to modulo dfplayer,  

Apparently works well the dfplayer fed directly 5v from the arduino, but e read that you have to feed from the battery 5v,  
That way it sounds louder or something like that.?

In the project that I'm using I'll use neopixel led strip, I must feed it to the same 5v connector of the led strip, or I still connect it with the 5v of the arduino.

Vulkan1, 2017-02-12:
ive done the DFplayer mini direct from the 5v of the arduino board output, no issues when you connect it properly, I was holding the connection with my fingers was realized I was getting static because of it.

shiryu55, 2017-02-13:
Nuclear tests here, neo pixel 60 led, http://www.leroymerlin.es/fp/16549743/tubo-multicapa-25-mm-
https://www.youtube.com/watch?v=3shLEjaUHxg

closed 2017-02-27 by shiryu55