neskweek/LightSaberOS #43: status led main button

shiryu55, 2017-02-17:
You could tell me where the arduino nano v3 pin can connect the status LED of the main button.

Protonerd, 2017-02-17:
Actually it can be any of the following A0-A3. You need a digital I/O to drive an LED, of course since the Ax pins are not PWM pins, you might need a series resistance to limit the current depending on the supply you use for the status/accent LED.

shiryu55, 2017-02-17:
Thank you for answering so quickly

In this link and hung a small scheme you could hit a small revision and correction.

[https://subefotos.com/ver/?71102316730e228d318a7949376c859ao.jpg#codigos](url)

shiryu55, 2017-02-26:
Connected the pin a0 with resistance, works well, to give the commands in the main led led of the sword.
So, find a solution for the led of the crystal chamber.