neskweek/LightSaberOS #46: test lightsaber os 1.5

shiryu55, 2017-02-26:
1.5 lightsaber loading attempt,  I use the arduino version 1.6.11

The skech loads well, without failure, but does not respond the arduino with the new program, does not emit start sound, nor does the power button work.

Protonerd, 2017-02-27:
I think I accidentally overwrote an earlier version of 1.5 with 1.3. Please let me fix this later today!

shiryu55, 2017-02-27:
No problem friend, one more thing when I run the link well, load the program with the version of arduino 1.6.11, it is correct??.... the ino, charge without problems but does not react to the power button, not sounds. is dead jjjj,.

The sd card has no file structure like the other version 1.4

carmonajeremias, 2017-03-14:
hi, alguien habla español?

shiryu55, 2017-03-15:
Yo mismo, pero idea poca, lo llevo no muy bien el proyecto, tengo algunas lagunas , pero si te puedo indicar algo cuenta conmigo.

carmonajeremias, 2017-03-15:
gracias shiryu. Tengo conocimientos de electrónica pero pocos de programación, arranque bajando todo lo de aca y pasándolo a arduino pero sin resultados positivos, También me doy cuenta que este sistema es muy avanzado para lo que necesito hacer.  Estoy armando un sable, con un led rgb de 3w, lo que necesito es simple, que prenda y apague que me detecte el swing y el fkash on clah, con sus respectivos sonidos. Acá me encuentro con menues configurables y miles de lineas de comando que me vuelven loco.  Vos me podrías dar una mano? Desde ya gracias por la respuesta sea favorable o no.

jbkuma, 2017-03-15:
RGB no funciona todavía en la versión 1.5.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

carmonajeremias, 2017-03-15:
Hola jbkuma, gracias por la respuesta, es que tampoco me funcionan los sonidos ni el swin. pero cuando los evaluo por separados andan sin problemas, Evidentemente tengo problemas al ensamblar el codigo.

Snyperdead, 2017-03-15:
con la versión 1.4 el RGB funciona sin problemas, al igual que el clash y el swing, si en algo te puedo apoyar con gusto. 

y en cuanto a la configuración de código realmente solo tienes que modificar algunas cosas no es nada complica en realidad cuando entiendes el concepto.

Saludos desde México.

shiryu55, 2017-03-15:
A mí con la versión 1,4 sin problema para rgb, o neopixel , en su momento mi fallo fue no utilizar la version 1.6.11  del programa , se ponen las librerías en su sitio, se comentan las líneas de código adecuadas en él config.h , para rgb en tu caso  , la posición del Mpu y poco más , anímate con las tiras neopixel están geniales , mis pruebas con la tira de 60 led metro no están mal , pero disimular los led es un poco problemático , estoy esperando las tiras de 120 led metro, mi correo es marioajasonido@gmail.con 
no consigo meter la 1.5 de protones me da un fallo a compilar. Utilizó la versión 1.16.11 con las librerías de este en el programa no sé si falta algo pero no compila .

También intente probar el código de jbkuna en su versión 1.4b pero no consigo que los sonidos funcionen en su orden.

Snyperdead, 2017-03-15:
Que tal Shiryu55, el problema del orden de los sonidos regularmente se debe a una mala colocación de los archivos en la Micro SD, formatea tu Micro SD y luego arrastra los archivos de audio a la tarjeta y deja que se copien. de esa manera se copian en el orden adecuado. si lo haces de alguna otra forma genera un orden aleatorio de los archivos y no en la secuencia correcta.

Saludos.

carmonajeremias, 2017-03-15:
Bueno, mucha gracias por todas las prontas respuestas, esta bueno saber que hay mucha gente de habla hispana en este foro. En este momento estoy desinstalando todo arduino para comenzar de cero. Shiryu respecto a neopixel esta buenisimo, pero no es lo que necesito, ya que mi sable lo voy a utilizar para practicar iaido, en el cual los sables se golpean, por eso necesito que no lleven nada adentro. La difucion lumínica la logre enrollando 3 hojas de papel celofán transparente dentro del tubo. en un rato armo todo de nuevo y les voy a realizar unas consultas mas. 

Muchas Gracias.

carmonajeremias, 2017-03-15:
el esquema de conexiones es el que esta dentro de la carpeta readme?

Snyperdead, 2017-03-15:
Que tal Carmonajeremias, buen día, 

Guíate con el siguiente link en el armado ensamblaje y configuración:

https://github.com/neskweek/LightSaberOS/tree/LSOS-1.4_singlebutton

ahí esta prácticamente todo, si necesitas mas apoyo avisas.

Saludos.

carmonajeremias, 2017-03-15:
chicos, respecto a los sonidos, dentro la carpeta "sd" hay 7 carpetas mas, "configsouns","jukebox" y "soundfont del 1 al 5". dentro de la tarjeta se copian todos los archivos en el root, o se copian con sus carpetas.?

shiryu55, 2017-03-15:
La forma de copiar los archivos es correcta, para la version 1.4 sin problemas , los copio como indica en el manual el problema es con la version 1.4b.

Con respecto a carmonajeremias,  ok si es para pegarte con el mejor el led en la base, cuantos metros de papel de celofán lo pusiste dentro , no me llega bien la luz hasta la punta del sable, y mira que utilice hasta led de 12w .

Este esquema me funciona para neo pixel y para led rgb, el color blanco no se como programarlo , en el codigo no tiene una salida para activar el color blanco, que tengo en mi led rgbw.

http://subefotos.com/ver/?5a1c62ac1ff9a705060a2cb4833bbdbfo.jpg

Snyperdead, 2017-03-15:
los sonidos Se copian tal cual están en el archivo en sus respectivas carpetas, y recuerda que primero debes formatear tu Micro SD y despues arrastar todos los archivos, de lo contrario se copian en forma aleatoria.
 
las "soundfont del 1 al 5" son los mismos archivos en cada carpeta, pero después los puedes reemplazar cuando encuentres o tengas alguna otra diferente a las que incluye el paquete.

carmonajeremias, 2017-03-15:
shiryu, tenes que poner en la punta algo que reflecte, intenta ponerle un pedazo de papel alumio solo en la punta para que reflecte hacia en interior. con eso deberia bastar.. Snyper Lo copie de esa forma. realice todo como dice en el manual, pero todavia no conecte led alguno. el swin sigue sin funcionar.

shiryu55, 2017-03-15:
esta puesto, pero no llega , jjjj, e probado con 3 tipos de led , y nada. por eso estoy con la tira neopixel
que raro que no suene , calibraste el mpu,  revisa las conexiones de la tarjeta del dfplayer.

Snyperdead, 2017-03-15:
cuando hace el boot tu arduino, escuchas el sonido de bienvenida??? o cuando presionas el Main Switch escuchar el encendido de saber?????

por otro lado, ya hiciste la calibración del  MPU 6050 y reemplazaste los datos en el config.h??????

también puede ser que tengas que configurar dentro del config.h la posicion o eje en el que tienes instalado tu MPU 6050

intenta golpear la base de tu protoboard o del MPU 6050 para ver si te da el sonido del clash.

carmonajeremias, 2017-03-15:
Shiryu. son aproxidamente 3 metros de celofan y en la punta aliminio reflectivo, use led de 3w sin problemas, despues subo imagen. Cuando arranca mi arduino (arduino nano) hace el hum directamente. Your offsets:	-1315	-1478	1125	38	-104	-21 los datos de calibracion del mpu. y cuando lo golpeo no tira el clash.

Ps: shiryu el blanco se logra con los 3 colores a su maxima potencia, unque le denota un toque azulado.

Snyperdead, 2017-03-15:
reemplazaste esos mismos datos dentro del Config.h
y que versión es la que instalaste en tu arduino?????

shiryu55, 2017-03-15:
Mira el diagrama chungo que colgue,  pero estoy con Snyperdead, creo que la posicion del mpu es lo que te esta tocando las narices.  es raro que no te suene el mensaje de entrada.
la jugada es que el led que tengo de 12w tiene el color de led blanco directo, pero en el codigo no da opción a ponerlo.

carmonajeremias, 2017-03-15:
instale la vercion 1.611.  no se si tiene algo que ver, pero todavia no puse ningun led.

Snyperdead, 2017-03-15:
me refiero a la version del LSOS, no a la version del IDE de arduino, y hasta donde yo se no es necesario tener conectado el led, para que haga la funsion, debería hacerlo sin necesidad de que este conectado.

shiryu55, 2017-03-15:
no pasa nada por no poner los led, y cuando pinchas en el menu de herramientas, placa  de arduino , gestor de tarjetas, tiene que estar seleccionado la version 1.6.11

shiryu55, 2017-03-15:
asi indica el manual

carmonajeremias, 2017-03-15:
shiryu uso tu diagrama de conexión? tiene un par de cosas cambiadas como el spk al arduino.

baje el LightSaberOS-LSOS-1.4_singlebutton.

shiryu55, 2017-03-15:
si funciona, perfectamente
lo que me queda es mirar como poner el led para el crystal chamber, nada mas

Snyperdead, 2017-03-15:
revisa la sigueinte linea en tu config.h y modificala acorde a la posicion de tu MPU 6050:

/*!!!!!IMPORTANT IMPORTANT IMPORTANT IMPORTANT IMPORTANT!!!
 *
 * MPU6050 device ORIENTATION
 * Choose which MPU's axis is parallel
 * to your blade axis
 *************************************/
//#define BLADE_X
#define BLADE_Y
//#define BLADE_Z
/************************************/

aunque deberia darte una bienvenida directamente al a ser el boot del arduino no hacer el humm directo (por la version)

carmonajeremias, 2017-03-15:
consulta. yo pase directamente las carpetas, las seleccione, las arrastre y las solte. esta bien, las carpetas con los archivos dentro.

Snyperdead, 2017-03-15:
es correcto carmo, en esta versión todo se maneja por carpetas, así como descomprimiste el archivo de audio se pasa directo a raiz de la Micro SD

carmonajeremias, 2017-03-15:
voy a desarmar todo y a repetirlo. en un rato les comento

carmonajeremias, 2017-03-15:
antes que nada. como no tengo conectado los leds. estoy alimentando por usb.

Snyperdead, 2017-03-15:
ohhhhhhhhh claroooooooooooooo

si esta aliemntando por USB probablemente es insuficiente, sobre todo para el df Player, te recomendaría que alimentaras el df player por separado, no con la salida de 5v del arduino.

y verifica bien la conexion del TX y RX igual del arduino al DF Player.

jbkuma, 2017-03-15:
Snyperdead, gracias por tu ayuda. Hablo español, pero sólo un poquito y no
durante muchos años. ¡Es difícil y agotador para mí! Es mucho mejor
comprender el idioma y el sujeto para explicarlo.  Parece que comprendes
LSOS muy bien, gracias por contribuir a la comunidad!

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

carmonajeremias, 2017-03-15:
Bueno chicos. arme todo de nuevo, cambie el arduino nano y dfplayer, ahora parece que anda. lastima que tengo a mi niña de 5 dias despierta y no lo puedo prender, en un rato le comento, pero parece que todo va sobre ruedas. Muchas gracias.

Snyperdead, 2017-03-15:
jbkuma:
Gracias, hago lo posible por apoyar cuando es necesario, comprendo el LSOS a prueba y error ademas de leer y hablar medianamente el ingles y ver los distintos problemas de otros usuarios en la comunidad.

Si necesitan apoyo y puedo ofrecerle (con el mucho o poco conocimiento que tengo de la plataforma) con gusto lo puedo hacer.

De igual manera estoy a disposición por si necesitas algo ya sea en español o ingles. 

Saludos.

Jbkuma:
Thanks, I do my best to support when it's necessary, I understand the LSOS on trial and error, in addition to reading and speaking medium English and seeing the different problems of other users in the community.

If they need support and I can offer you (with much or little knowledge I have of the platform) I can gladly do it.

I am also available if you need anything in Spanish or English.

Greetings.

Snyperdead, 2017-03-15:
carmona jeremias, espero que todo salga bien y que la info que te hemos dado te pueda ayudar o te haya ayudado para echar a volar tu LSOS, si necesitas algo mas y te puedo apoyar buscame y trataremos de resolverlo.

por cierto felicidades por tu pequeña y disfrutala al maximo, yo tengo una de 2 y es mi adoración aunque no me deje trabajar en los hobbies jajaajajajajaja!!!!!

Saludos.

shiryu55, 2017-03-15:
Esto como dices es ensayo y error , y gracias a compañeros como los que estáis comentando en este post e podido disfrutar de la construcción de mi sable , todavía queda por pulir cosas y molestaros un poco pero la cosa va funcionando, y poder comentar en español es un mundo de diferencia

carmonajeremias, 2017-03-15:
Snyp que cosa los hijos son la luz que nuestros ojos, yo tengo otra de tres que estaba en el jardin. Chicos funciono todo de maravillas, despues capaz los moleste con las conecciones de leds. Shiryu te dejo el link, esta hoja mide 80cm, tiene 2mts de celofan transparente en el interior sujetada en el extremo interior con silicona caliente y en la parte superior aluminio. Lo que si hay que ponerle es una lupa (colimador led) que hace que la luz del led se refracte en 15° o menos, esto es importantisimo que lo tenga. http://articulo.mercadolibre.com.ar/MLA-629279115-lente-lupa-colimador-led-alta-potencia-1w-3w-5w-15-grados-_JM ahi te dejo la descripcion.  https://youtu.be/OVfn4vsPnS8 este es el video de la hoja. 

Gracias a todosssssssssssssssssssssssssssssssss

Snyperdead, 2017-03-15:
para que el led funsione adecuadamente es necesario el lente adecuado para focalizar la emision de la luz,
son como estos 

http://electronica.com.ve/new/catalog/images/lente-para-led-luxeon-60-grados.jpg

y en cuanto a los grados espero les sirvan esta imagenes 

http://i01.i.aliimg.com/img/pb/061/655/825/825655061_735.jpg

http://www.faciled.es/images/Angulos-de-vision.jpg

shiryu55, 2017-03-15:
Tengo la lente y pese a todo no propaga muy bien La Luz , es muy tenue .

Realizaré más pruebas nucleares este fin de semana, me llegan las tiras de led neopixel de 120led y más material.
http://www.slimsaber.com/slimsaber/es/electronica/26-led-12w-xml-rgbw.html
Este es el último led q e estado probando

Snyperdead, 2017-03-15:
mmmm raro!

los leds cree deberían darte suficiente potencia y luminosidad para abarcar la vaina del sable sin mayores problemas, tal vez deberías intentar con otro tipo de difusor, tal vez policarbonato blanco translucido como core, eso te podría ayudar a dar el efecto y propagar la luz del led hasta el Final de tu vaina.

shiryu55, 2017-03-15:
claro aquí son las 21.40 de la noche, jjjjj, este año no podre ir de vaciones para mexico, llevo 6 años seguidos pasando mis vacas allí.

shiryu55, 2017-03-15:
para agosto me toca ser papi, de un nene. se me jodió el enredar , asi que ando aprobechando el tiempo que resta.........

carmonajeremias, 2017-03-15:
Felicitacionesssssssssssssssss por el nene.

shiryu55, 2017-03-15:
Ya te dire cuando me desquicie del todo jjjjj,,,

acabo de compilar la version 1.5 y subida al arduino sin problemas.

perooooo...... al meter el calibration.ino  me pega este fallo
Arduino:1.6.11 (Windows 7), Tarjeta:"Arduino Nano, ATmega328"

Arduino:1.6.11 (Windows 7), Tarjeta:"Arduino Nano, ATmega328"

MPU6050_calibration:23: error: stray '\302' in program

   <title>DIYino/MPU6050_calibration.ino at master Â·
Protonerd/DIYino</title>

   ^

MPU6050_calibration:23: error: stray '\267' in program

MPU6050_calibration:114: error: missing terminating ' character

   </option></form><form
accept-charset="UTF-8" action="/Protonerd/DIYino/search"
class="js-site-search-form"
data-scoped-search-url="/Protonerd/DIYino/search"
data-unscoped-search-url="/search" method="get"><div
style="margin:0;padding:0;display:inline"><input name="utf8" type="hidden"
value="&#x2713;" /></div>

   ^

MPU6050_calibration:230: error: missing terminating ' character

         </option></form><form
accept-charset="UTF-8" action="/logout" class="logout-form"
method="post"><div style="margin:0;padding:0;display:inline"><input
name="utf8" type="hidden" value="&#x2713;" /><input
name="authenticity_token" type="hidden"
value="cn4vGdaoBdkPVAu1kfPvivQ6CJNTk6kcMCgtuqMeLEQLv7VgrN+IiuatcHE8LJj28GIyaRk2wLdFIPh2F7msqg=="
/></div>

         ^

MPU6050_calibration:240: error: missing terminating ' character

     </option></form><form
accept-charset="UTF-8" action="/logout" class="sr-only right-0"
method="post"><div style="margin:0;padding:0;display:inline"><input
name="utf8" type="hidden" value="&#x2713;" /><input
name="authenticity_token" type="hidden"
value="tFZsIcwL7poVFkBnj1kpc06VV3uGzCpk14rSLVibu2HNl/ZYtnxjyfzvO6Mihl4PSs1tgcxpQ8+iggfh7Dw7jw=="
/></div>

     ^

MPU6050_calibration:271: error: missing terminating ' character

         </option></form><form
accept-charset="UTF-8" action="/notifications/subscribe"
class="js-social-container" data-autosubmit="true" data-remote="true"
method="post"><div style="margin:0;padding:0;display:inline"><input
name="utf8" type="hidden" value="&#x2713;" /><input
name="authenticity_token" type="hidden"
value="wEho09CDzbz97CjYcxraiJUnzKvBraOcZcnV4STwPVZZRnZtHfcGs62JInQd+wZjqlqgQo47tC2LVAFQPOQcnA=="
/></div>      <input class="form-control" id="repository_id"
name="repository_id" type="hidden" value="60167880" />

         ^

MPU6050_calibration:302: error: stray '@' in program

[... 800 lines cut ...]

                                                 ^

MPU6050_calibration:893: error: expected unqualified-id before '<' token

         <td id="LC66" class="blob-code blob-code-inner js-file-line"><span
class="pl-k">unsigned</span> <span class="pl-k">int</span>
calibratedOffsetAdress = <span class="pl-c1">0</span>;</td>

                                     ^

MPU6050_calibration:902: error: expected unqualified-id before '<' token

         <td id="LC68" class="blob-code blob-code-inner js-file-line"><span
class="pl-k">unsigned</span> <span class="pl-k">int</span> loopcount=<span
class="pl-c1">0</span>;</td>

                      ^

exit status 1
stray '\302' in program

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

Snyperdead, 2017-03-15:
probablemente te falte un punto y coma en algún lugar del sketch, cierra tu ide vuelve a cargar tu sketch o descarga solamente el sketch de calibración y mételo nuevamente pareciera que solo es un error de dedo.

shiryu55, 2017-03-15:
lo e probado unas cuantas veces, y mismo resultado y en diferentes pc

Snyperdead, 2017-03-15:
mmmmm me puedes mandar tu sketch y así le doy una revisada o por acá tengo algunas versiones del sketch de calibración en mi maquina, solo tengo que encontrarlo entre tantos archivos jejejejejeje, avísenme a donde se los mando o igual luego formamos algún grupo de whatsapp o algo así para que sea mas fácil.

o pensándolo bien, tal vez sea la librería del MPU 6050, recuerdo que Protonert había hecho algunas modificaciones pero no estoy seguro. así que carga una librería nueva de mpu 6050, no la del LSoS e intenta calibrarlo.

Saludos

shiryu55, 2017-03-16:
jjjjj,,eso me temo todos mis amigos me dicen lo mismo , aprobecha a dormir,,,,,,,,,,

 estoy intentando seguir el tutorial, con lo que utilizo el script recomendado 

https://github.com/Protonerd/DIYino/blob/master/MPU6050_calibration.ino , pero este no compila , las librerias que tengo son las que estan en el propio arduino 1.6.11 mas las que trae el propio lightsaber 0s en su version 1.5

jbkuma, 2017-03-16:
Debe configurar el script.
//#define MEMORYBASEMPUCALIBOFFSET 96 // start storing the calibration
values starting at this address <-- para LSOS 1.4 y los antes
#define MEMORYBASEMPUCALIBOFFSET 200 // start storing the calibration
values starting at this address <-- para LSOS 1.5

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

shiryu55, 2017-03-16:
Muchas gracias kuma como siempre , lo mirare esta noche , no estaría de más para la comunidad en español un tutoríal paso a paso para los que tienen conocimientos limitados de arduino y electrónica , y ganas de conseguir tener un sable láser

shiryu55, 2017-03-16:
Por mi parte si se realiza un tutorial paso a paso con fotos y video , de todo el proceso , ayudare en lo que pueda a la creación del mismo.
aqui expongo unas fotos del ultimo prototipo
http://subefotos.com/ver/?71f23318548ced07a93c688aa97b1f93o.jpg
http://subefotos.com/ver/?fd6d214b1ba773970bbffcdee2ca269co.jpg

aqui un video interesante de la creacion de un sable laser.
https://www.youtube.com/watch?v=AMn91xJ-KW0&t=215s

Un saludo a todos y muchísimas gracias por la atención y dedicación a este hobby.

Snyperdead, 2017-03-16:
Seria bastante bueno, trabajar en una version de LSoS en español sobre todo documentacion jejejejjeje.

Shiryu, se ve vastante bien tu prototipo, pero me surge una pequeña duda, por que has usado dos resistencias smd 471 ( 470 ohms )?????

Saludos.

shiryu55, 2017-03-16:
Para conectar el pin de datos el 13 para la tira de led neopixel 

El tutoríal , me mandas un correo electrónico y te mando un PRE tutoríal y rellenas lo que me falte a mi

carmonajeremias, 2017-03-16:
buenas chicossssssss, me mande un moco, empece a jugar con la programacion del sable, con el boton, y al parecer le baje el volumen, y ya no se escucha, le cargue de nuevo el soft, pero tampoco. ayudaaaaaaaa, jajajaja

Snyperdead, 2017-03-16:
Que versión tienes cargada antes que todo jejejejej?????

carmonajeremias, 2017-03-16:
jaja ya easta, ahora solo me quedo bajito, la 1.4

Snyperdead, 2017-03-16:
Puedes hacer 3 cosas, 

1.-Jugar a atinarle al menú, que si no mal recuerdo es el primero,esperar unos segundo e ir aumentando hasta llegar al volumen deseado, si te pasas pues sigues dando click hasta un punto antes ya que no tengo aquí en la oficina mi arduino, 

2.- puedes conectar el arduino y en el IDE activar el Monitor Serial e ir viendo a cada click la acción que se registra en pantalla y ajustar ahí mismo el nivel de volumen

3.- ejecutar un sketch de limpieza de EEProm para limpiar correctamente tu arduino y volver a cargar el Sketch del LSoS nuevamente.

[60 more replies]