neskweek/LightSaberOS #47: Atmega328p

alessiotb, 2017-03-03:
Hi, this is not a issue, but i'd like to ask you if it's possible run this sketch on a atmega328p stand alone, because i just tried and it didn't worked, do you know if there are some issues with the internal oscillator at 8MHz?

jbkuma, 2017-03-03:
What sort of problem are you having?

alessiotb, 2017-03-04:
It's quite hard to explain, i just loaded it and...nothing worked ^^" 
the power supply was ok, cause the green led on mpu6050 was lit, and the dfplayer, if i short to ground the 10th pin, it plays the humming sound, probably i loaded the wrong *.hex, because i was on linux and i porbably loaded the wrong one, when i manage to buy a new usb typeB i'll try again!