neskweek/LightSaberOS #48: не могу разобратся с ошибкой

DRagonenko, 2017-03-14:
[image]

Protonerd, 2017-03-14:
Could you please translate to English? We are not that good in Russian...

jbkuma, 2017-03-14:
It looks like you might have a non-standard version of I2Cdev.  The
standard library doesn't ask for a em_cmu.h, line 37 is part of the
copyright notice.
here is the official library
https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/I2Cdev

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

DRagonenko, 2017-03-14:
when you run the compile errorArduino: 1.6.13 Hourly Build 2016/11/11 10:33 (Windows 7), Плата:"Arduino Nano, ATmega328" In file included from C:\Users\Р?С‡РµРіРѕ\Desktop\LightSaberOS\LightSaberOS.ino:15:0: C:\Users\Ичего\Documents\Arduino\libraries\I2Cdev/I2Cdev.h:37:20: fatal error: em_cmu.h: No such file or directory  #include <em_cmu.h>                     ^ compilation terminated. exit status 1Ошибка компиляции для платы Arduino Nano.

DRagonenko, 2017-03-14:
Please share your version of the program with the necessary Arduino libraries that do not throw errors

faeton13, 2017-03-14:
Задавать  вопрос на английском портале в англоязычном заберу на русском несколько самоуверенно как мне кажется. Имя пользователя на кириллице идея того же порядка. В этой строке ничего не кажется подозрительным? " C:\Users\Р?С‡РµРіРѕ\Desktop\LightSaberOS...."

OK, have your read  "readme.MD" file?

DRagonenko, 2017-03-14:
in the end, the problem has not changed

DRagonenko, 2017-03-15:
in the end I removed the Arduino and all its components . re-installed Arduino and the libraries and booted

faeton13, 2017-03-15:
И как? Помогло?

DRagonenko, 2017-03-16:
helped

DRagonenko, 2017-03-16:
you fellows have created such beauty