neskweek/LightSaberOS #49: Hellow, Help.  someone speaks Spanish?

carmonajeremias, 2017-03-14:
I need help with my saber, but y dont speack english.

Snyperdead, 2017-03-15:
yo hablo español, cual es el problema.

Saludos desde México.

leofaure, 2017-04-01:
Hola yo hablo español. por fin encuentro gente de habla hispana con mi interés.Estoy creando mi primer lightsaber con mucho entusiasmo y paciencia, nunca antes havia hecho nada paracido y estoy aprendiendo mucho al respecto pero sigo muy verde. mi proyecto es a base de una tira de led casera y un arduino nano , mpu5060 y dfplayer, tengo todo cableado menos la tira le ded.Hoy he decidido usar la versión 1.5 a esperas que alguien me de una mejor, por otra parte el archivo de audios esta perfecto, es algo que faltaba en otras versiones. acabo de conectarlo todo y al encenderlo ha sonado ¨diynolightsaber¨ asi que parece esta bien pero me falta saber el tema de los botones ya que los tengo puestos asi: una pata a tierra y la otra a sus correspondientes pines. el 12 y el 4, no estoy seguro si devo darle corriente a ese pin o no.Me alegra conocer mas hispanohablantes espero que seamos mas para ayudarnos ya que suelo usar traducor de google y es muy tedioso y poco exacto.
un saludo!