neskweek/LightSaberOS #50: LUXEON CODE error

Vulkan1, 2017-03-17:
Hi,

so I went to update from 1.4 to 1.5, I modified the config so it was the same as 1.4

I get this error, a new one for me, and I cant even comprehend what it means

Arduino: 1.6.13 (Windows 8.1), Board: "Arduino Nano, ATmega328"

```
exit status 1
cannot convert 'cRGB' to 'uint8_t {aka unsigned char}' for argument '2' to 'void getColor(uint8_t*, uint8_t)'
```

Any ideas
im fairly sure its related to LUXEON, due to this being the error line (the second line)
` #if defined LUXEON
          getColor(currentColor, storage.sndProfile[storage.soundFont].blasterboltColor);`

Protonerd, 2017-03-17:
Hi Vulkan. it is unfortunately known that LUXEON is not compiling, sole reason is that LUXEON was not in focus for the 1.5. I also do not have any LUXEON build, therefore I did the verification only for neopixels and led strings. I need volunteers to debug the code for LUXEON until I manage to swap the board to DIYino in one of my LUXEON type sabers. Care to debug? The thing is, I wanted to use this cRGB type not only for neopixels to store the color setting but also generally for all RGB LEDs (incl. LUXEON). But the LUXEON functions still are not updated to work with the new type.

Vulkan1, 2017-03-20:
I'm currently undergoing full time study, but I have have a go every now and then

Vulkan1, 2017-03-20:
Also which neo pixels are you using, mind sending me a link?

Protonerd, 2017-03-23:
Yes, I'm using these ones: http://www.ebay.de/itm/1m-30-60-144-RGB-LED-Strip-Stripe-mit-WS2812B-5050-SMD-LEDs-WS2812-Versand-ZJ-/172344003101?var=&hash=item282080ca1d

This is a German ebay site, but you get the idea I guess. Important is the 144LED/m, WS2812B controller type

Vulkan1, 2017-03-23:
I thought in the code it said max 120. Or with the memory saved in the 1.5v can you use many more

Protonerd, 2017-03-23:
I strove to achieve 120 because that many will fit into a ~32" blade, which is more or less the standard blade length. And less than 1m :)
probably the code can be optimized to accommodate more than 120, I never tried to find the exact limit, 120 is good enough. You need to consider that with 2 stipes back to back, you end up with 2x120=240, in terms of current draw that is already running the boards to the extreme limits.

Vulkan1, 2017-03-23:
ill order some and start playing around with them, not a bad price for the Chinese bands

Protonerd, 2017-03-24:
I fixed the LUXEON (not STAR_LED) code and just committed the updates to v1.5, try and feed back!

Vulkan1, 2017-03-24:
Will do

jbkuma, 2017-03-24:
I guess I can stop working on it ;)

jbkuma, 2017-03-25:
It looks like this merges in my 1.5.3 code, does it include the fixes for
single button?
Lines 511 and 513 in the main .ino should be changed from BUTTONLEDPIN to
ACCENT_LED
All of the accent functions are handled via the ACCENT_LED pin and
BUTTONLEDPIN should be eliminated since it not used and the configured pin
shares with the FTDI/MP3 switches (depending on the board).

Protonerd, 2017-03-25:
Yes, indeed it is based on your code, I fixed single button and STAR_LED as well. I will look into the BUTTONLEDPIN/ACCENT_LED controversy... thanks for pointing  it out!

jbkuma, 2017-03-25:
I'm pretty sure the only reference to BUTTONLEDPIN is in my "quick mute"
code.  Since I hadn't been using an accent LED I didn't look closely enough
until it didn't work.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Protonerd, 2017-03-25:
Vulkan, is it working now for you? It works with LUXEOn on my side...

Vulkan1, 2017-03-25:
Sorry been busy with work and uni, that said I'm going to hook up my setup tonight and run it off some leds. 
I built mine modular so my main advantage is. I can just pull off the speaker, buttons, battery and plug them into the other main "motherboard". Working on a neopixel one in eagle. Software is blowing my mind right now.

Protonerd, 2017-03-26:
If you want to have a reliable solution, go for a DIYino board.