neskweek/LightSaberOS #51: Sharp buzz after playing menu sound files

772pilot, 2017-03-20:
I've encountered an issue where there a sharp buzz will play from the speaker immediately after a non-saber sound has played. This includes boot sounds, menu navigation files, and jukebox files. I've attached a video demonstrating the bug. 

Video of issue (30 second clip): [https://youtu.be/0jRbk4iWHkA](https://youtu.be/0jRbk4iWHkA)

I'm running a slightly modified version of LSOS 1.3 (only lighting is modified) in a luxeon configuration.  Hardware is standard: chinese clone nano, MPU6050, and DFPlayer. I'm also using the Arduino IDE (ver. 1.8.1).

It's worth noting that this issue only popped up after I resolved a previous issue where no sound was being played AT ALL (DFPlayer red indicator LED was not lighting up). The Arduino IDE compiler told me there was a conflict between multiple instances of DFPlayer.h so I renamed the "DFPlayer" folder to "DFPlayerOLD" and elected to use "DFPlayer_LSOS" as the only library. It was only after this that I noticed the buzzing. As a sanity check, I removed both the "DFPlayerOLD" and "DFPlayer_Mini_Mp3" folders from the libraries folder, but that didn't change anything.

Any ideas as to what could be causing the issue and how to fix it?

jbkuma, 2017-03-20:
It sounds like what happens if there is meta data in the file.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Vulkan1, 2017-03-20:
I get that as well but less noticeable. Mine is ignorable but I believe it's clicks more or less with mine.

772pilot, 2017-03-20:
Wow, it didn't even occur to me it could be the metadata. Everything works perfectly after stripping it all, thanks!

closed 2017-03-20 by 772pilot