neskweek/LightSaberOS #52: Config Menu Wrong Audio Being Played When Switching Menus After Modification

772pilot, 2017-03-20:
As the title says, I've encountered a config mode bug in which the min / max audio files are played when changing menus after a modification has been made to the previous menu. 

EXAMPLE:
-I'm currently in the "VOL" menu and my current volume is 15. 
-I increase the volume +5 to 20 by clicking the main button five times, but I have not hit the maximum volume yet (30). Ergo, the "max" audio does not trigger. 
-I change menus by holding the main button. 
-The LEDs change to indicate I am now in "blade color" menu, but the female announcer audio for "blade color" does not play. Instead I hear either the "min" or "max" sounds (tracks 10 and 11 in the CONFIG folder). 
-This only happens if I modify the settings within each menu and not when I skip directly from menu to menu. (i.e., it happens when I'm modifying volume or something and NOT when I switch from SNDFNT->VOL->BLADE COLOR)

I've verified that all my files have been copied over to the SD card correctly (audio is fine everywhere else), and that Soundfont.h reflects the contents of my SD card.

Running on LSOS 1.3, chinese nano clone, MPU6050, and DFPlayer. Using Arduino IDE v1.8.1 with board manager 1.6.11.

Any thoughts?

Protonerd, 2017-03-23:
Did you try v1.5 as well? It could be that the config files have changed since v1.3? I guess yes.

772pilot, 2017-03-25:
I haven't yet. I cannot compile v1.5. I'm getting this error: 
LightSaberOS:317: error: 'class MPU6050' has no member named 'dmpInitialize_light'

Vulkan1, 2017-03-25:
As for the dmpinitialise_light. Make sur ethe libary is up to date. I had that issue as well. I tried commenting it but I got no swing sounds (https://github.com/neskweek/LightSaberOS/issues/41)
Give that a go

Protonerd, 2017-03-25:
:) RTFM . Most common mistake, you haven't updated your libraries. Starting from 1.4 (or even 1.3?) I made the MPU init code leaner, probably you still have thd old lib.

Protonerd, 2017-03-25:
Thanks Vulkan, you came first!

772pilot, 2017-03-25:
This is all happening after installing the new libraries.

Vulkan1, 2017-03-25:
Make sure you have got the right directory. Same issue as me, I was updating the wrong library's.

Vulkan1, 2017-03-25:
"If you go in preferences and turn on "verbose output" you may see something
like this:

Multiple libraries were found for "Wire.h"
 Used: C:\[...]\packages\arduino\hardware\avr\1.6.17\libraries\Wire
 Not used: D:\[...]\Documents\Arduino\libraries\Wire

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe"

Vulkan1, 2017-03-25:
I don't know how to quote, sorry Jason.

772pilot, 2017-03-25:
Yeah, I'm seeing that. For DFPlayer as well, actually.

Vulkan1, 2017-03-25:
The avr and wire were examples, u need them. The mpu6050 libary is still out of date. What version are you getting your libary idea from? And what version are you running.

As for the DFPlayer this is the same. I believe in 1.5 the library's have been edited to accomodate new features. That said this would be a question for proto. But that's what I believe is going on.

772pilot, 2017-03-25:
I'm using the library that comes with the LSOS GravityBlend v1.5 master download. Same with DFPlayer.

Vulkan1, 2017-03-25:
Did you install all the new libraries from the folder without a capital L

jbkuma, 2017-03-25:
Our next release will indeed have some new features and a thoroughly
scrubbed code base.  Some of the features have been incorporated into the
latest update of 1.5

You can see the progress here:
http://jbkuma.com/workshoppe/307/lsos1-5-2jb/

There are 5 new config sounds for the battery menu, you can get the
additional files here:
https://github.com/jbkuma/LightSaberOS/blob/LSOS-1.5.3/ConfigSounds.zip
(this git is my personal build which is heavily altered and some things may
not work for you)

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

772pilot, 2017-03-25:
Oh interesting. The "libraries" folder is not included in the ZIP download, only the "Libraries" folder.

772pilot, 2017-03-26:
Ok, I installed the new libraries, but I'm still getting this:

```
In function 'void SleepModeEntry()':

LightSaberOS:1452: error: 'class DFPlayer' has no member named 'sleep'

   dfplayer.sleep();

```

I'm definitely using the appropriate library so I'm confused now.

Vulkan1, 2017-03-26:
We have elmininated most of the errors which is good. This one is a question for one of the developers. that being proto or anyone else.

772pilot, 2017-03-26:
If it helps, DFPlayer.h in the "libraries" folder seems to be exactly the same as the regular "Libraries" folder. Perhaps the updated version is not reflected on github?

jbkuma, 2017-03-26:
I'm getting the same error. You can disable deep sleep mode in config.h or
comment the line for now, this only affects sleep mode which mainly
pertains to those using DIYino boards.

Protonerd, 2017-03-26:
I now made the 2 libraries (L and l) identical, both should have the right library versions. Sorry for the confusion. It was actually a copy/paste problem, when I made the 1.5 branch, it created the original files from the master. When I overwrote the files, I just selected all files needed for a full package. Unfortunately I did not recognize that in my local drive library was with lowercase, therefore it created out of sheer case sensitivty :) a new lib folder. You might ask yourself why I choose not to remove the bad folder: it is very difficult or time consuming (either one of them depending on how you wish to do it) to delete files from a Git,,,

772pilot, 2017-03-27:
Got the new library installed and switched to Eclipse Arduino. Getting the following compilation errors now:

```
Arduino\libraries\DFPlayer_LSOS/DFPlayer.h:647:2: error: 'DFPlayerSerial' does not name a type
  DFPlayerSerial * mp3Serial;
```

```
Arduino\libraries\DFPlayer_LSOS/DFPlayer.h:128:15: error: 'mp3Serial' was not declared in this scope
   if (NULL != mp3Serial) {
```

```
Arduino\libraries\DFPlayer_LSOS/DFPlayer.h:129:17: error: 'class DFPlayer' has no member named 'mp3Serial'
    delete this->mp3Serial;
```

Protonerd, 2017-03-27:
We are not supporting Eclipse. Please use Arduino IDE. Report if something is not working there.

772pilot, 2017-03-28:
Ah, I understand. I compiled it in Arduino IDE and it worked just fine. However, the menu on LSOS 1.5 seems to be buggier than 1.3. When in the volume menu, holding the main button does not skip to the next menu option, but rather increases the volume another two ticks. This is far removed from my initial problem.

Vulkan1, 2017-03-28:
U need to disable single button in config if you havent

772pilot, 2017-03-28:
I have.

Vulkan1, 2017-03-28:
Have you ruled out that it's not your switch. Get two jumper cables (one to ground and one to the main button port) hold the open wire against each other simulating a button hold.
I had a bad button and it did similar things, triggered twice when you press rather then hold. This could be the issue.

772pilot, 2017-03-28:
Yup, I've verified the button works fine. My LSOS v1.3 still works as it should (except for the original bug).

Protonerd, 2017-03-28:
No :) that 2 beeps are a signal that the board is ready to go into sleep mode! THat is a new feature in fact. If you do not want to have it, try commenting out 'define DEEP_SLEEP'

Vulkan1, 2017-03-28:
This is interesting, I'll have to play with 1.5 just been swamped with uni work.

772pilot, 2017-03-29:
Ah, gotcha Proto. I got 1.5 running on my nano and the original bug does not seem to be present. Guess I'll just port my modifications to 1.5!