neskweek/LightSaberOS #53: please help Mpu 6058

DRagonenko, 2017-03-23:
Could you help me with Mpu 6058. in project LSOS 1.4, please? It works only by hit. And movement works only with a hit of the player or doesn't work at all. I can't understand how to fix it. I had coded colibration data but nothing changed.
Sorry if I have mistakes. I'm not good enough in English

Protonerd, 2017-03-23:
Hi DRagonenko,
If you are using the MPU6050, please first execute the calibration using the sketch: https://github.com/Protonerd/DIYino/blob/master/MPU6050_calibration.ino

Here please take care to select the right position in the EEPROM to store the values. If you use LSOS v1.5, please use:
#define MEMORYBASEMPUCALIBOFFSET 200

If an earlier version:
#define MEMORYBASEMPUCALIBOFFSET 96

In the LSOS code the calib values will be fetched from this location, code-wise it can be found here:
#ifdef MPUCALOFFSETEEPROM
        // retreive MPU6050 calibrated offset values from EEPROM
        EEPROM.setMemPool(MEMORYBASEMPUCALIBOFFSET, EEPROMSizeATmega328);
        int addressInt=MEMORYBASEMPUCALIBOFFSET;
       mpu.setXAccelOffset(EEPROM.readInt(addressInt));
        #ifdef LS_INFO
          int16_t output;
          output = EEPROM.readInt(addressInt);
          Serial.print("address: ");Serial.println(addressInt);Serial.print("output: ");Serial.println(output);Serial.println("");
        #endif
         addressInt = addressInt + 2; //EEPROM.getAddress(sizeof(int));        
       mpu.setYAccelOffset(EEPROM.readInt(addressInt));
        #ifdef LS_INFO
          output = EEPROM.readInt(addressInt);
          Serial.print("address: ");Serial.println(addressInt);Serial.print("output: ");Serial.println(output);Serial.println("");
        #endif
         addressInt = addressInt + 2; //EEPROM.getAddress(sizeof(int));
       mpu.setZAccelOffset(EEPROM.readInt(addressInt));
        #ifdef LS_INFO
          output = EEPROM.readInt(addressInt);
          Serial.print("address: ");Serial.println(addressInt);Serial.print("output: ");Serial.println(output);Serial.println("");
        #endif
         addressInt = addressInt + 2; //EEPROM.getAddress(sizeof(int));
       mpu.setXGyroOffset(EEPROM.readInt(addressInt));
        #ifdef LS_INFO
          output = EEPROM.readInt(addressInt);
          Serial.print("address: ");Serial.println(addressInt);Serial.print("output: ");Serial.println(output);Serial.println("");
        #endif
        addressInt = addressInt + 2; //EEPROM.getAddress(sizeof(int));
        mpu.setYGyroOffset(EEPROM.readInt(addressInt));
        #ifdef LS_INFO
          output = EEPROM.readInt(addressInt);
          Serial.print("address: ");Serial.println(addressInt);Serial.print("output: ");Serial.println(output);Serial.println("");
        #endif
         addressInt = addressInt + 2; //EEPROM.getAddress(sizeof(int));
       mpu.setZGyroOffset(EEPROM.readInt(addressInt));
        #ifdef LS_INFO
          output = EEPROM.readInt(addressInt);
          Serial.print("address: ");Serial.println(addressInt);Serial.print("output: ");Serial.println(output);Serial.println("");
        #endif
 #else // assign calibrated offset values here:
 /* UNIT1 */
	mpu.setXAccelOffset(46);
	mpu.setYAccelOffset(-4942);
	mpu.setZAccelOffset(4721);
	mpu.setXGyroOffset(23);
	mpu.setYGyroOffset(-11);
	mpu.setZGyroOffset(44);
#endif

Vulkan1, 2017-03-23:
Have a look at this, it sounds like the problem I had.

Vulkan1, 2017-03-23:
https://github.com/neskweek/LightSaberOS/issues/41

Forgot the link

DRagonenko, 2017-03-24:
did the calibration
[image]

amended
[image]
uploaded

and the same thing
please help

Protonerd, 2017-03-24:
you have to amend it in the Config.h file like this:
#define MPUCALOFFSETEEPROM
#ifdef MPUCALOFFSETEEPROM
#define MEMORYBASEMPUCALIBOFFSET 200 // 96 -> increased due to storing RGB PWM values in EEPROM takes a lot of space (3 x 1byte x 3 color types x nr of sound fonts i.e. for 5 sound fonts 45 bytes
#endif

DRagonenko, 2017-03-24:
[image]

DRagonenko, 2017-03-24:
only blows =(

Protonerd, 2017-03-24:
When you calibrated the board with the sketch I've sent you, did you press y and enter at the end to save the values into the EEPROM? Looking at the screen shot I have the impression that you did not.

DRagonenko, 2017-03-25:
pressed . =(

DRagonenko, 2017-03-25:
tried everything. Only the collision
the dream is dead =(((
please help

Protonerd, 2017-03-25:
OK, please send me your whole wiring diagram. Or buy a DIYino board, those boards are assured to work! If I know your circuitry, I might figure out what is wrong. BTW, what do you mean by "collosion"?

DRagonenko, 2017-03-26:
https://yadi.sk/d/bI9wkqg63GLyHq

"collosion"(works only by hit)
Sorry if I have mistakes. I'm not good enough in English

DRagonenko, 2017-03-27:
Thanks for the help. updated libraries and seems to work. thank you very much

DRagonenko, 2017-04-10:
when driving hangs playing music