neskweek/LightSaberOS #54: A few warnings on compile

CobraMoses, 2017-03-25:
Hello,

I'm new to Arduino and saw this project and built it.  I'm running 1.6.11 on an UNO and initially saw a mass amount of errors when i compiled until i updated LinkedList.h.  Now i have the following left.. any ideas?

C:\Users\Moses\Desktop\LightSaberOS\LightSaberOS.ino: In function 'void loop()':

C:\Users\Moses\Desktop\LightSaberOS\LightSaberOS.ino:540:34: warning: iteration 3u invokes undefined behavior [-Waggressive-loop-optimizations]

     digitalWrite(ledPins[i], HIGH);

                                  ^

C:\Users\Moses\Desktop\LightSaberOS\LightSaberOS.ino:539:4: note: containing loop

    for (uint8_t i = 0; i <= 5; i++) {

Protonerd, 2017-03-25:
Which branch do you use? And when did you last update it?

CobraMoses, 2017-03-25:
These errors were from 1.4.  That was after not having any success compiling 1.3.  Not sure what i'm doing wrong.

Protonerd, 2017-03-26:
Did you try 1.5? Did you also install the libraries?
Which Arduino IDE version do you use and which board definition? As to what these terms mean, please refer to the readme file.

CobraMoses, 2017-03-26:
I just tried 1.5 and got the errors below.  I had previously moved over the libraries from the folder to the default sketch libraries folder in 'Documents'.  As stated in original post, i am using 1.6.11 and i am using the 1.6.17 as per the readme file.  Tried with both a Nano and Uno in settings.

C:\Users\Moses\Desktop\LightSaberOS\LightSaberOS.ino: In function 'void SleepModeEntry()':

LightSaberOS:1452: error: 'class DFPlayer' has no member named 'sleep'

   dfplayer.sleep();

            ^

Multiple libraries were found for "Wire.h"
 Used: C:\Users\Moses\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.17\libraries\Wire
 Not used: C:\Users\Moses\Documents\Arduino\libraries\Wire
exit status 1
'class DFPlayer' has no member named 'sleep'

DRagonenko, 2017-03-30:
completely remove Arduino. and all libraries . then re-install and alternately install the libraries which the program asks for
#48
Sorry if I have mistakes. I'm not good enough in English

Protonerd, 2017-03-30:
And does it work now?

DRagonenko, 2017-03-31:
included . movement catches . but sometimes freezes movement and begins to fail

DRagonenko, 2017-03-31:
works for me

CobraMoses, 2017-03-31:
I'll try this tonight and close this thread if it works for me

leofaure, 2017-03-31:
hi all, i have some error, 'class DFPlayer' has no member named 'sleep'.
before i cleaned and uninstall and reinstall all but ...

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/leo/Documents/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10802 -build-path /var/folders/10/jxq8rmhs33j7kvyclnpqjyj40000gn/T/arduino_build_922585 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/leo/Documents/Arduino/LightSaberOS/LightSaberOS.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/leo/Documents/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10802 -build-path /var/folders/10/jxq8rmhs33j7kvyclnpqjyj40000gn/T/arduino_build_922585 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/leo/Documents/Arduino/LightSaberOS/LightSaberOS.ino
Using board 'nano' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "/var/folders/10/jxq8rmhs33j7kvyclnpqjyj40000gn/T/arduino_build_922585/sketch/LightSaberOS.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "-I/Users/leo/Documents/Arduino/libraries/DFPlayer_LSOS" "/var/folders/10/jxq8rmhs33j7kvyclnpqjyj40000gn/T/arduino_build_922585/sketch/LightSaberOS.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "-I/Users/leo/Documents/Arduino/libraries/DFPlayer_LSOS" "-I/Users/leo/Documents/Arduino/libraries/I2Cdev" "/var/folders/10/jxq8rmhs33j7kvyclnpqjyj40000gn/T/arduino_build_922585/sketch/LightSaberOS.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "-I/Users/leo/Documents/Arduino/libraries/DFPlayer_LSOS" "-I/Users/leo/Documents/Arduino/libraries/I2Cdev" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "/var/folders/10/jxq8rmhs33j7kvyclnpqjyj40000gn/T/arduino_build_922585/sketch/LightSaberOS.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "-I/Users/leo/Documents/Arduino/libraries/DFPlayer_LSOS" "-I/Users/leo/Documents/Arduin [... cut ...]
Utilizando biblioteca EEPROMEx en carpeta: /Users/leo/Documents/Arduino/libraries/EEPROMEx (legacy)
Utilizando biblioteca OneButton en carpeta: /Users/leo/Documents/Arduino/libraries/OneButton (legacy)
Usando librería LinkedList con versión 1.2.2 en la carpeta: /Users/leo/Documents/Arduino/libraries/LinkedList 
Utilizando biblioteca light_WS2812 en carpeta: /Users/leo/Documents/Arduino/libraries/light_WS2812 (legacy)
exit status 1
'class DFPlayer' has no member named 'sleep'

any idea?
 thanks for all

jbkuma, 2017-03-31:
You can comment the line to get it to compile.  This is part of the Deep
Sleep function.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

leofaure, 2017-03-31:
If I done it after writing the post and compiled it, but I'm not sure if it worked normally and correctly.
I want it.
Can anyone give a schematic about the main and aux buttons? I doubt if you add resistance to the buttons or not? I am not clear connections.
Another way anybody can give a scheme about the main and aux buttons? I doubt if you add resistance to the buttons or not? I am not clear connections.

Thanks for all.

some

CobraMoses, 2017-03-31:
Thanks for the help Protonerd (and lovely initiative to do this software), and Спасибо DRagonenko .. i followed what  you said and am now able to compile.

closed 2017-03-31 by CobraMoses

jbkuma, 2017-04-01:
Wiring diagrams are available here:
https://github.com/Protonerd/FX-SaberOS/wiki/Wiring

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

leofaure, 2017-04-01:
Amazing thanks so much! , any help about the error ?'class DFPlayer' has no member named 'sleep', for moment i comment this phrase.