neskweek/LightSaberOS #55: Error compiling

dannyc09, 2017-04-05:
Getting this issue when trying to compile:

In file included from Buttons.cpp:8:
C:\Users\Dan and Heather\Documents\Arduino\libraries\DFPlayer/DFPlayer.h: In member function 'void DFPlayer::receive()':
C:\Users\Dan and Heather\Documents\Arduino\libraries\DFPlayer/DFPlayer.h:278: error: invalid conversion from 'uint8_t*' to 'char*'
C:\Users\Dan and Heather\Documents\Arduino\libraries\DFPlayer/DFPlayer.h:278: error:   initializing argument 2 of 'size_t Stream::readBytesUntil(char, char*, size_t)'

Thanks
Dan

Protonerd, 2017-04-05:
which LSOS version are you using? Please note that the master is already obsolete for some time now. Use one of the braches, best is 1.5

dannyc09, 2017-04-05:
Using 1.5 I get the following error code on compile:

In file included from LightSaberOS.cpp:45:
C:\Users\Dan and Heather\Documents\Arduino\libraries\DFPlayer_LSOS/DFPlayer.h: In member function 'void DFPlayer::receive()':
C:\Users\Dan and Heather\Documents\Arduino\libraries\DFPlayer_LSOS/DFPlayer.h:279: error: invalid conversion from 'uint8_t*' to 'char*'
C:\Users\Dan and Heather\Documents\Arduino\libraries\DFPlayer_LSOS/DFPlayer.h:279: error:   initializing argument 2 of 'size_t Stream::readBytesUntil(char, char*, size_t)'
In file included from LightSaberOS.cpp:53:
C:\Users\Dan and Heather\Documents\Arduino\libraries\OneButton/OneButton.h: At global scope:
C:\Users\Dan and Heather\Documents\Arduino\libraries\OneButton/OneButton.h:59: error: ISO C++ forbids initialization of member '_debounceTicks'
C:\Users\Dan and Heather\Documents\Arduino\libraries\OneButton/OneButton.h:59: error: making '_debounceTicks' static
LightSaberOS.cpp: In function 'void setup()':
LightSaberOS:238: error: 'loadConfig' was not declared in this scope
LightSaberOS:283: error: 'saveConfig' was not declared in this scope
LightSaberOS:517: error: 'InitDFPlayer' was not declared in this scope
LightSaberOS:522: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.cpp: In function 'void loop()':
LightSaberOS:573: error: 'HumRelaunch' was not declared in this scope
LightSaberOS:594: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS:614: error: 'motionEngine' was not declared in this scope
LightSaberOS:627: error: 'motionEngine' was not declared in this scope
LightSaberOS:647: error: 'LoopPlay_Sound' was not declared in this scope
LightSaberOS:661: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS:702: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS:917: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS:942: error: 'HumRelaunch' was not declared in this scope
LightSaberOS:972: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS:986: error: 'batCheck' was not declared in this scope
LightSaberOS:1051: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS:1077: error: 'saveConfig' was not declared in this scope
LightSaberOS:1089: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS:1127: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS:1163: error: 'SleepModeEntry' was not declared in this scope
LightSaberOS:1168: error: 'SleepModeExit' was not declared in this scope
LightSaberOS:1172: error: 'SinglePlay_Sound' was not declared in this scope
LightSaberOS.cpp: In function 'void HumRelaunch()':
LightSaberOS:1339: error: 'LoopPlay_Sound' was not declared in this scope
LightSaberOS.cpp: In function 'void SleepModeEntry()':
LightSaberOS:1451: error: 'class DFPlayer' has no member named 'sleep'

Protonerd, 2017-04-05:
Did you read the instructions?
Some hints:

please use Arduino IDE version 1.6.11 or newer
make sure you use the Board Manager version 1.6.17 or newer!!! How to do this?
go to Tools/Boards/Boards Manager...
select Arduino AVR Boards by Arduino version xxxx
Select in the lower left side of the box the version 1.6.17 and press install
if you want to use the Gravity Color Blend feature, you need to move the MPU6050 calibration values so make room for the extended set of arameters stored in the EEPROM. The easierst way to do it is to re-calibrate the MU using this sketch (MPU lib needed!!!): https://github.com/Protonerd/DIYino/blob/master/MPU6050_calibration.ino

dannyc09, 2017-04-06:
Looks like I had the wrong version of the IDE...though I had updated.

Now getting the following errors:

Arduino: 1.8.2 (Windows 10), Board: "Arduino Nano, ATmega328"

In file included from sketch\Buttons.cpp:10:0:

sketch\Config.h:263:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\Config.h:274:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST 

       ^

sketch\Config.h:332:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\Buttons.cpp:810:17: warning: extra tokens at end of #else directive

           #else if STAR_LED

                 ^

sketch\Buttons.cpp:823:17: warning: extra tokens at end of #else directive

           #else if STAR_LED

                 ^

sketch\Buttons.cpp:835:17: warning: extra tokens at end of #else directive

           #else if STAR_LED

                 ^

sketch\Buttons.cpp:1102:17: warning: extra tokens at end of #else directive

           #else if STAR_LED

                 ^

sketch\Buttons.cpp:1114:17: warning: extra tokens at end of #else directive

           #else if STAR_LED

                 ^

sketch\Buttons.cpp:1126:17: warning: extra tokens at end of #else directive

           #else if STAR_LED

                 ^

In file included from sketch\SoundFont.h:12:0,

                 from sketch\Buttons.cpp:11:

C:\Users\Dan and Heather\Documents\Arduino\libraries\LinkedList/LinkedList.h: In instantiation of 'LinkedList<T>::LinkedList() [with T = unsigned int]':

sketch\SoundFont.h:30:14:   required from here

[... 182 lines cut ...]
   tmp->next = false;

             ^

C:\Users\Dan and Heather\Documents\Arduino\libraries\LinkedList/LinkedList.h:256:8: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]

   root = false;

        ^

C:\Users\Dan and Heather\Documents\Arduino\libraries\LinkedList/LinkedList.h:257:8: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]

   last = false;

        ^

Multiple libraries were found for "Wire.h"
 Used: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire
 Not used: C:\Users\Dan and Heather\Documents\Arduino\libraries\Wire
exit status 1
'class DFPlayer' has no member named 'sleep'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dannyc09, 2017-04-06:
I've uploaded the calibration code but nothing happens in the serial monitor...

Protonerd, 2017-04-06:
Did you set the right baud rate? Should be 110k (from the top of my head). What about the previous error? Does it still persist?

dannyc09, 2017-04-07:
I set the baud rate to 115200.  The serial monitor still doesn't pick anything up.

I've tried the DIYino test sketch and the serial monitor picks it up but sticks at "Initializing I2C devices"

Protonerd, 2017-04-07:
Are you using a DIYino board? If yes, the test sketch must work. If not, please check your connections. BTW, in the LSOS code most of the seerial messages are disabled with compile directives not to take precious code space. So now it only brings mostly one single line and then nothing more.

dannyc09, 2017-04-07:
Yes I'm using the DIYino board.

Some good news...I've managed to get the MPU configuration running - it said the test failed in the end but asked me to save values to EPPROM.  I'm assuming this is correct?

I've then run the DIYino test sketch and the serial monitor now loads and runs a script but is constantly running showing a range of acceleration and gyro values.  Is this correct?

When trying to upload LSOS 1.5 I get this:

Arduino: 1.8.2 (Windows 10), Board: "Arduino Nano, ATmega328"

C:\Users\Dan and Heather\Documents\Arduino\LightSaberOS\LightSaberOS.ino: In function 'void SleepModeEntry()':

LightSaberOS:1452: error: 'class DFPlayer' has no member named 'sleep'

   dfplayer.sleep();

            ^

exit status 1
'class DFPlayer' has no member named 'sleep'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jbkuma, 2017-04-07:
if you comment the line it should work fine.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

DRagonenko, 2017-04-11:
LSOS-1.5
[image]
Arduino: 1.8.2 (Windows 7), Плата:"Arduino Nano, ATmega328"

C:\Users\Р?С‡РµРіРѕ\Desktop\LightSaberOS\LightSaberOS.ino: In function 'void SleepModeEntry()':

LightSaberOS:1452: error: 'class DFPlayer' has no member named 'sleep'

Несколько библиотек найдено для "Wire.h"
 Используется: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire
Не используется: C:\Users\Ичего\Documents\Arduino\libraries\Wire
exit status 1
'class DFPlayer' has no member named 'sleep'

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

faeton13, 2017-04-11:
Comment that line and all similar to it

Protonerd, 2017-04-11:
Hi dannyC, does it work now? What you described about the MPU calibration and the DIYino Testsketch is OK.
I do not understand the sleep error message...AFAIK I run the same code and no trouble. Anyway, this sleep command does not leave any impression on the DFPlayer chip, so it's safe to comment it out. On my Prime v1.5 and Stardust I simply cut power to the module.

Protonerd, 2017-04-11:
On suggestion from jbkuma I removed the troublesome sleep command. Still not sure what causes it, probably a library mismatch.

dannyc09, 2017-04-13:
Ok...apologies for the delay.

Finally got it to compile and upload..whoop!

On the downside I managed to fry my board and ruined the contacts for connecting the string blade.  My own fault so it looks like I'll be ordering a new board.

On a side note I did find that the sounds worked a bit weirdly:

-  The "hum" sound came on when power was run to the board even before the blade was lit.

- When moving the DIYino around there was no change to the sound to replicate motion or swinging.

I will try again with a fresh board and see if that makes any difference.  I've learnt from my mistakes here so fingers crossed my v2 attempt will be more successful.