neskweek/LightSaberOS #57: error compiling star-led 1.5

shiryu55, 2017-04-16:
Fault when compiling, the fault occurs when you comment the single button line, to have the functions in a single button.

Arduino:1.8.1 (Windows 7), Tarjeta:"Arduino Nano, ATmega328"

sketch\Buttons.cpp: In function 'void mainDoubleClick()':

Buttons.cpp:350: error: too many arguments to function 'void getColor(cRGB)'

             getColor(currentColor, storage.sndProfile[storage.soundFont].mainColor);

                                                                                   ^

In file included from sketch\Buttons.cpp:12:0:

sketch\Light.h:68:6: note: declared here

 void getColor(cRGB color); //getColor

      ^

Buttons.cpp:379: error: too many arguments to function 'void getColor(cRGB)'

             getColor(currentColor, storage.sndProfile[storage.soundFont].mainColor);

                                                                                   ^

In file included from sketch\Buttons.cpp:12:0:

sketch\Light.h:68:6: note: declared here

 void getColor(cRGB color); //getColor

      ^

Buttons.cpp:402: error: too many arguments to function 'void getColor(cRGB)'

             getColor(currentColor, storage.sndProfile[storage.soundFont].mainColor);

                                                                                   ^

In file included from sketch\Buttons.cpp:12:0:

sketch\Light.h:68:6: note: declared here

 void getColor(cRGB color); //getColor

      ^

Buttons.cpp:451: error: too many arguments to function 'void getColor(cRGB)'

           getColor(currentColor, storage.sndProfile[storage.soundFont].clashColor);

                                                                                  ^

In file included from sketch\Buttons.cpp:12:0:

sketch\Light.h:68:6: note: declared here

 void getColor(cRGB color); //getColor

      ^

Buttons.cpp:474: error: too many arguments to function 'void getColor(cRGB)'

           getColor(currentColor, storage.sndProfile[storage.soundFont].blasterboltColor);

                                                                                        ^

In file included from sketch\Buttons.cpp:12:0:

sketch\Light.h:68:6: note: declared here

 void getColor(cRGB color); //getColor

      ^

Se encontraron múltiples librerías para "Wire.h"
Usado: C:\lightsaber 1.5n\arduino-1.8.1\hardware\arduino\avr\libraries\Wire
 No usado: C:\lightsaber 1.5n\arduino-1.8.1\libraries\Wire
exit status 1
too many arguments to function 'void getColor(cRGB)'

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

Protonerd, 2017-04-16:
There has been soooo many updates to v1.5, it is very popular :) I know that the first upload only supported neopixels and string blades. I know I have a working STAR_LED code since I just finished a STAR_LED build, my very first one. But I'm not sure even the current v1.5 fully compiles for STAR_LED. I need to check... but I remember these errors, I had the same, I needed to prune through the whole code to make it work. As you might know from the history, LSOS was meant to be a software mainly for led strings and neopixel stripes, and while we planned supporting STAR_LED all the while, progress on it was slow at best due to none of the developers having a STAR_LED build. Now I have one with my Stardust :) Prod me with requests and as soon as I have a code I'm hay with I will send it to you.

shiryu55, 2017-04-18:
Thank you very much for your work. I have seen in the code a new section to look at the state of the battery, with a couple more connections and resistances, 
Another section where the type of plaque is chosen if it is home made or original ..

I am getting mad, with the theme of the fet, which smd fet me you recommend for the project.

http://subefotos.com/ver/?71f23318548ced07a93c688aa97b1f93o.jpg
http://subefotos.com/ver/?fd6d214b1ba773970bbffcdee2ca269co.jpg

In other case, we are creating a little forum with the telegram for Spanish-speaking people. Creating a tutorial to solve the small doubts and do not mess with nonsense.

https://t.me/joinchat/AAAAAAyyXTH4MGfOzYYGIQ