neskweek/LightSaberOS #59: compile error TCCr2a was not declared in this scope

Basprojects, 2017-05-01:
Hi, 

Just downloaded the lightsaberOS library and program, but when I just open the LightsaberOS.ino file in Arduino 1.6.11 and press verify, it will give me the error TCCR2A was not declared in this scope.

Any idea's ???

jbkuma, 2017-05-01:
TCCR2A is a hardware register for the PWM.  There shouldn't be any scope
issues.  What hardware are you using, and what IDE board manager version
are you using?

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Basprojects, 2017-05-01:
Hi,

Thanks for the quick response ;-).

I'm using a leonardo, but it allready goes wrong on the verify in the arduino IDE. The version is 1.6.12 as mentioned, or do you mean some another thing by the IDE board manager??

Br,

Sebastiaan

Verstuurd vanaf mijn iPhone

http://jbkuma.com/workshoppe

jbkuma, 2017-05-01:
Board manager is a separate item.

The problem is that the current version of LSOS has some items specific to
the ATmega328 based boards (DIYino, Nano, ProMini, Uno), and the Leonardo
is 32u4.  This is in the lightOff() function for string blades in Light.cpp

The are definitely ways you can adjust the code to work with Leonardo, but
I don't think it's a priority for anyone at the moment.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Basprojects, 2017-05-03:
Hi.

I have changed the board to an atmeg328 version and the compiling has gone well.. 

I have tried to understand the whole config.h file, as I understand you must set your preferences on the kind of led strips etc.. In my case as I only have the arduino board and a neopixel strip and some buttons, I only want to set the config.h to neopixel output and the main switch for on/off and a secondary color switch, which is defined, right? The sound and movement I don't want to use at this moment. Is this at all possible only by commenting the right options in the config.h file??  As I tried this but then I get compilling errors in the saberos.ino

Thanks for the help so far, appreciate it! ;-)

jbkuma, 2017-05-03:
I don't think you'll be able to disable the movement without a lot of extra
work. The pixel code should work really well as that has been out major
focus.

Basprojects, 2017-05-18:
Hi,

After receiving all the items I needed to do the built I managed to get everything working with neopixel and a arduino pro mini! It works great and I'm really happy with it 👍 

There is one issue though, if I long press the main switch to turn off the blade, it switches off, but afterwards it does not respond on anything anymore... I have to perform a complete reset on the arduino board or disconnect the power supply to make it work again... Any reasons why it would do that??

closed 2017-06-28 by Basprojects