neskweek/LightSaberOS #61: How can disable the grativy function

shiryu55, 2017-06-02:
I just do not like the option, and I would like to know how to disable this function in this revision of lsos1.5

DRagonenko, 2017-06-09:
[image]
raskomentiruj

shiryu55, 2017-06-10:
I can understand that the photo indicates, it is to reduce the number of colors, I do not understand that I have to do to deactivate the gravity function

What I want to activate  the color change as in version 1.4

DRagonenko, 2017-06-10:
raskomentiruj
и попробуй

Protonerd, 2017-06-11:
The arrows point in the right direction, if COLORS is defined, Gravity color mix is disabled and you can selectt again from a set of pre-defined colors. BTW, I will stick to the Gravity color mix as it's way mor advanced and allows a wide range of color hues to be defined rather than selecting from a list, starting from the next version. One more reason is that besides being way better Gravity color mix is also lighter on memory.