neskweek/LightSaberOS #62: Jukebox Playing on Launch

Vulkan1, 2017-06-18:
Hi,
ive been crunching at this all day, and really haven't gotten anywhere.
Im running a home made Arduino Lightsaber, using the schematic provided on the read me page.
Ive defined it as a homebrew.

Once the lightsaber has launched it will play the jukebox.
Pressing the main button only switches between my first song, and then gets stuck on the "Title Scene" Song. Pressing the secondary button does nothing. I cant understand why this is, it was working fairly well in 1.3, but has crashed down in 1.5

Thanks for your help in advance.

faeton13, 2017-06-18:
I have same problem, cause by "shortcut" for secondary button. Single button option should hide that problem.
I have in mine case problem was from error in power wiring.

jbkuma, 2017-06-18:
This is not a software issue, it is a setup issue.  1.3 and 1.5 have
different eeprom addresses and sound files, you can't just load 1.5.  Make sure
the config is correct, your eeprom is properly set, and your SD card
matches Soundfont.h

Protonerd, 2017-06-18:
What jbkuma wrote is true, I migh add that AFAIK the homebrew is an empty define with no content behind it. I included it for any future usage, but currently if you select it, all it does is skip code which might well be the cause of your troubles. Specify Prime and feed back if it gets better.

Vulkan1, 2017-06-19:
Okey so I switched back to the prime. 
The only issues that persist now is that:
when entering config, instead of jumping to volume, it plays the title scene, sound does not increase when you click main or aux, but the beeps occur. To get past this one hold the main button and cycle though the menus. You eventually get back around to volume and it works.

When changing volume the lit neopixels representing sound level (which I love btw, its fantastic), it eventually get to a limit, where the red is no longer shown and the green pixel is barely it (a speckle of green), now I have interpreted that I have gone beyond my limit of sound level, or its playing up.

Relating to above, if those green speckles are shown, after exiting config and start using the saber as per normal, sound will eventually cut out.

Hope the information helps, if you need anything else I will provide.
I love your work guys, please keep it up, its amazing the joy your are bringing to people!