neskweek/LightSaberOS #63: After switching off light saber it won't power on anymore with main switch only complete reset

Basprojects, 2017-06-28:


Protonerd, 2017-10-12:
How many neopixels are you trying to use? Sometimes if you are using too many, the board will simply run out of dynamic memory and the code will end up in nirvana. That is why currently a limitation of max 120 pixels is recommended.

Basprojects, 2017-10-30:
Hi, 

Thanks for the reply.

I only have 13 neopixels attached... I allready tried several boards now all with the same outcome. Also when looking at the serial monitor, after pressing long main switch nothing happens anymore.

I tried to search in the config.h file and the button.h file but did not find any strange code there. 

Maybe you could sent me an .INO file setup for 13 neopixels that should work.. Maybe I'm overlooking something really tiny/

Thanks anyway.

DnkMemes, 2018-09-03:
Hello,

I am having the same issue and was hoping to check in and see if anyone has figured anything out. Amy help would be greatly appreciated.

Thank you

Protonerd, 2018-09-03:
This Git is already old and not any more maintained. For a hugely improved code go to FX- SaberOS.