neskweek/LightSaberOS #64: Blaster blocage !

Tam-Koowan, 2017-06-28:
Hello, pourrais tu crée un effet d'arrêt de blaster sur la lame à des endroit aléatoire sur la lame lié aux sons d'arrêt de blaster  ??

jbkuma, 2017-06-28:
This is exactly how it works with the pixel blade.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Tam-Koowan, 2017-06-29:
Supe; j'ai plus qu'à commander une nano.
Savez-vous si on peux faie ce montage avec une arduino pro micro
http://tinkersphere.com/1398-thickbox_default/pro-micro-5v-16mhz-arduino-compatible-atmega32u4-breakout.jpg
?

Merci

Tam Koowan Props

jbkuma, 2017-06-29:
> 16mhz-arduino-compatible-atmega32u4-breakout.jpg

Any ATmega328 based board should work

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Tam-Koowan, 2017-07-09:
J'ai acheter une Nano V3 mais une erreur de compilation, alors que j'ai tous mis dans le bon dossier  !!
[image]